Xinyuan-LilyGO/LilyGo-LoRa-Series #4: Error

robstermn, 2019-12-31:
Can't compile, I'm getting the following error?

exit status 1
'class LoRaClass' has no member named 'setPins'

lewisxhe, 2020-01-02:
Click button 2 will cycle the motherboard function, the default start to enable GPS positioning, the second click will be set to lora send, the third click will be set to lora receive, long press for two seconds and then release, will enter deep sleep

robstermn, 2020-01-02:
I finally got it working with an arduino IDE, but I can't seem to get it to compile with VSCode/PlatformIO?  I'm getting the following errors:

`
class "LoRaClass" has no member "setPins"
class "LoRaClass" has no member "begin"
class "LoRaClass" has no member "beginPacket"
class "LoRaClass" has no member "print"
class "LoRaClass" has no member "endPacket"
class "LoRaClass" has no member "parsePacket"
class "LoRaClass" has no member "available"
class "LoRaClass" has no member "read"
class "LoRaClass" has no member "packetRssi"
class "LoRaClass" has no member "packetRssi"

`

closed 2020-12-04 by lewisxhe