Xinyuan-LilyGO/LilyGo-LoRa-Series #7: no matching function for call to 'Button2::Button2()' error when compiling

Damien-Abot, 2020-01-22:
Hello,

I'm getting this error when trying to compile the code in arduino for my new TTGO t22_v1.0 20190612 version board.
`home/damien/Bureau/TTGO-T-Beam/TTGO-T-Beam.ino: In function 'void button_init()':
TTGO-T-Beam:61:30: error: no matching function for call to 'Button2::Button2()'
     pBtns = new Button2 [args];

                              ^
In file included from /home/damien/Bureau/TTGO-T-Beam/TTGO-T-Beam.ino:5:0:
/home/damien/Arduino/libraries/Button2-master/src/Button2.h:55:5: note: candidate: Button2::Button2(byte, byte, unsigned int)
     Button2(byte attachTo, byte buttonMode = INPUT_PULLUP, unsigned int debounceTimeout = DEBOUNCE_MS);
     ^
/home/damien/Arduino/libraries/Button2-master/src/Button2.h:55:5: note:   candidate expects 3 arguments, 0 provided
/home/damien/Arduino/libraries/Button2-master/src/Button2.h:29:7: note: candidate: constexpr Button2::Button2(const Button2&)
 class Button2 {
       ^
/home/damien/Arduino/libraries/Button2-master/src/Button2.h:29:7: note:   candidate expects 1 argument, 0 provided
/home/damien/Arduino/libraries/Button2-master/src/Button2.h:29:7: note: candidate: constexpr Button2::Button2(Button2&&)
/home/damien/Arduino/libraries/Button2-master/src/Button2.h:29:7: note:   candidate expects 1 argument, 0 provided
Plusieurs bibliothèque trouvées pour "LoRa.h"
Utilisé : /home/damien/Arduino/libraries/LoRa
Plusieurs bibliothèque trouvées pour "axp20x.h"
Utilisé : /home/damien/Arduino/libraries/axp20x
Plusieurs bibliothèque trouvées pour "SSD1306.h"
Utilisé : /home/damien/Arduino/libraries/esp8266-oled-ssd1306-cb294b0d5e938864ba935a8e917082e5496399ca
Plusieurs bibliothèque trouvées pour "Wire.h"
Utilisé : /home/damien/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/Wire
Plusieurs bibliothèque trouvées pour "SPI.h"
Utilisé : /home/damien/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/SPI
Plusieurs bibliothèque trouvées pour "TinyGPS++.h"
Utilisé : /home/damien/Arduino/libraries/TinyGPSPlus-1.0.2b
Plusieurs bibliothèque trouvées pour "WiFi.h"
Utilisé : /home/damien/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi
Non utilisé : /home/damien/softwares/arduino-1.8.10/libraries/WiFi
Plusieurs bibliothèque trouvées pour "Button2.h"
Utilisé : /home/damien/Arduino/libraries/Button2-master
Plusieurs bibliothèque trouvées pour "Ticker.h"
Utilisé : /home/damien/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/Ticker
exit status 1
no matching function for call to 'Button2::Button2()'
`
I installe the button2 library from this address
`https://codeload.github.com/Xinyuan-LilyGO/TTGO-T-Beam/zip/master`

Could you please help me ?
Thanks,

lewisxhe, 2020-01-24:
This is the changed Button2 @Damien-Abot 
https://github.com/lewisxhe/Button2/tree/fed9e73972c8c3436dde5fe56d98ac7b7e7e9d41

Damien-Abot, 2020-01-24:
Thank you @lewisxhe,

Works great with this version of library.

closed 2020-01-24 by Damien-Abot