Xinyuan-LilyGO/LilyGo-LoRa-Series #8: T-BEAM enters deep sleep after 0.5s of power on, does not flash

invidtiv, 2020-02-04:
T-Beam when powered on by usb , no battery , lights up the ssd1606 display for half a second showing the first screen from the demo , and then powers off.
No light turn on the board at any time.
With Liion on the board, nothing ever happens.
Cannot flash any more error always shows up.

Any direction of what can be done now to be able to flash the T-BEAM with another sketch.'?
Is there anyway to reset the power module?

invidtiv, 2020-02-04:
I m was using this TTGO-T-Beam.ino

lewisxhe, 2020-02-07:
Use USB power supply to display the first screen after booting, then shut down, can't boot anymore?

invidtiv, 2020-02-14:
No there was a problem with the sleep , I managed to reset the powermodule and was therefore able to upload code again.

closed 2020-02-14 by invidtiv