Xinyuan-LilyGO/LilyGo-LoRa-Series #9: Long battery life

SeungRyeol, 2020-02-06:
I configured the Sender function with only two GPS and LoRa TTGO-T-Beams. I want to use lithium-ion 2600mAh to guarantee at least a week of operating time. GPS transmission cycle is 1 second, Device status update is 10 seconds about LoRa.

How can I solve it?

Best Regards,

lewisxhe, 2020-02-07:
A cycle of 10 seconds. I want to achieve a week's battery life. I find it a bit difficult. Can I extend the sending cycle?
If you can quickly locate it every time you start it, then it will improve the working cycle. According to me, if the GPS is under the condition of maintaining the RTC power and the signal is good, the startup to the end of the positioning is within 10 seconds, which means that the device wakes up Positioning, plus the sending period, is in the range of about 10 seconds, but this is only theoretical and may actually take more time

SeungRyeol, 2020-02-07:
Please make sure I understand your comment.

1.Change the GPS reception period to 10 seconds,
2. Turn off GPS while waiting for next reception using axp.setPowerOutPut (AXP192_LDO3, AXP202_OFF)
= Turn off the GPS power whenever I receive GPS data once every 10 seconds.
(Because you use RTC)

lewisxhe, 2020-02-07:
No, I suggest you extend the positioning cycle
Turn off LDO2, LDO3 after each transmission is completed, and set esp32 to deep sleep

SeungRyeol, 2020-02-07:
you mean
```
#define uS_TO_S_FACTOR 1000000  // Conversion factor for micro seconds to seconds
#define TIME_TO_SLEEP  10        // gps data receiving cycle

function(); // gps data receiving & lora sender function
axp.setPowerOutPut (AXP192_LDO2, AXP202_OFF); // for lora
axp.setPowerOutPut (AXP192_LDO3, AXP202_OFF); // for gps
esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR); // deep sleep
```

Did I understand your opinion?

lewisxhe, 2020-02-07:
Yes . But I think 10 seconds is a bit short, look at yourself.

SeungRyeol, 2020-02-07:
okay, I will try it. 

Thank you very much for your advice.

closed 2020-02-07 by SeungRyeol

baarratia, 2021-04-28:
> I configured the Sender function with only two GPS and LoRa TTGO-T-Beams. I want to use lithium-ion 2600mAh to guarantee at least a week of operating time. GPS transmission cycle is 1 second, Device status update is 10 seconds about LoRa.
> 
> How can I solve it?
> 
> Best Regards,

Hi, did you finally achieved the week long battery life? I'm working towards a similar objective and It would be amazing if you can give me some of your insights.

Thanks!