Xinyuan-LilyGO/LilyGo-LoRa-Series #10: Data reception problems

SeungRyeol, 2020-02-07:
I am using T-beam v1.0 to receive data. Using the basic example, we received a LoRa packet, but after a certain period of time we did not receive the data (Sender keeps sending data). So I've modified the code a bit, but after a while, the receiver stops receiving packets. The board is working fine, but I want to know why it stops receiving packets.

* origin code
Same as the example in this repo.

* modified code
```
void lora_init(){
#ifdef ENABLE_LOAR
    SPI.begin(LORA_SCK, LORA_MISO, LORA_MOSI, LORA_SS);
    LoRa.setPins(LORA_SS, LORA_RST, LORA_DI0);
    if (!LoRa.begin(BAND))
        Serial.println("LORA Begin FAIL");
    else {
        loraBeginOK = true;
        Serial.println("LORA Begin PASS");
    }

    LoRa.setSpreadingFactor(LORA_SF);
    LoRa.setCodingRate4(LORA_CR);
    LoRa.setSignalBandwidth(LORA_BW);
    LoRa.setPreambleLength(LORA_PREAMBLE_LENGTH);
    LoRa.enableCrc();
    
    LoRa.onReceive(onReceive);// register the receive callback
    
    /* Set LoRa RX-mode */
    LoRa.disableInvertIQ();               // normal mode
    LoRa.receive();                       // set receive mode
#endif
}

void onReceive(int packetSize) {
    xTaskCreate(getPacket, "getPacket", 10240, (void*)&packetSize, CRONJOB_READLORA_PRIORITY, NULL); // thread
}

void getPacket(void* pvParameters){
    int packetSize = *((int*)pvParameters);
    String recv = "";
    int index = 0;
    char *token = NULL;

    // read packet
    for (int i = 0; i < packetSize; i++) {
        recv += (char)LoRa.read();
    }

    Serial.println(recv);

    // Cleanup task before exiting
    vTaskDelay(1);
    vTaskDelete(NULL);
}

void loop(){
}

```

lewisxhe, 2020-02-07:
You set the receive callback to create a task. I do n’t know what you are doing for this, and this is a thread-unsafe method. You can try adding a thread lock or not creating a task.

SeungRyeol, 2020-02-07:
Thank you for the reply! The above code(using thread) is written because of the problem of stopping reception during operation. The original code is as follows:
```
void lora_init()
{
#ifdef ENABLE_LOAR
    SPI.begin(LORA_SCK, LORA_MISO, LORA_MOSI, LORA_SS);
    LoRa.setPins(LORA_SS, LORA_RST, LORA_DI0);
    if (!LoRa.begin(BAND))
        Serial.println("LORA Begin FAIL");
    else {
        loraBeginOK = true;
        Serial.println("LORA Begin PASS");
    }
#endif
}

void loop(){
    if (axp192_found && pmu_irq) {
        pmu_irq = false;
        axp.readIRQ();
        if (axp.isChargingIRQ()) {
            baChStatus = "Charging";
        } else {
            baChStatus = "No Charging";
        }
        if (axp.isVbusRemoveIRQ()) {
            baChStatus = "No Charging";
        }
        digitalWrite(2, !digitalRead(2));
        axp.clearIRQ();
    }

    if (!loraBeginOK) {
        recv =  "Lora Begin FAIL";
        return;
    }
    if (LoRa.parsePacket()) {
        recv = "";
        while (LoRa.available()) {
            recv += (char)LoRa.read();
        }
        Serial.println(recv);
    }
}
```
I don't know why it works well and stops after a certain amount of time.

lewisxhe, 2020-02-07:
Working for a while, how long is this time?

SeungRyeol, 2020-02-07:
It will work for about 5-7 hours(using origin example code) or 20-23 hours(using modified code - thread). The time counter works(the board is alive) but can't receive LoRa data. Sometimes I receive data for more than 20 hours, but have the same problem.

lewisxhe, 2020-02-07:
Wow . I have to check this. We haven't worked yet. I will test it after work to see if there is this problem.

Do you send continuously or regularly?

SeungRyeol, 2020-02-07:
That's right. Receive GPS information by one or two devices once a second.

SeungRyeol, 2020-02-26:
Every time I received a lora packet, I noticed a decrease in heap memory.

You can check with the esp_get_free_heap_size () function.

As you will see further testing, I think it will no longer receive Lora packets when the heap area memory is low(= near 0).

closed 2020-03-02 by SeungRyeol