Xinyuan-LilyGO/LilyGo-LoRa-Series #11: TP54001819 on T22_V06 20180705 burning

FLSCHI, 2020-02-18:
Hello,
when powering the T22_V06 20180705 with an 18650 battery which got charged in an external battery charger to exactly 4.16V will destroy the TTGO-Beam! 
The TP54001819 chip (responsible for charging?) on the TTGO-Beam board will become extremely hot until smoke rises and bye-bye-beam! (happend on 2 devices!)
Is this issue solved in later revisions or will this still be the same behaviour? 
Thanks,
FS

lewisxhe, 2020-02-19:
V6 is an old product. When did you buy it?

FLSCHI, 2020-02-19:
I think roundabout 1.5 years ago. 

Will this also happen with a later revision?

lewisxhe, 2020-02-26:
There are two versions behind, one is TP5400 and the other is AXP192, which uses two different power chips

peterwishart, 2020-10-27:
I have a similar problem with a T22_V06 which had been sitting for a while, powered off but with battery installed. Battery was down to 3.8v.
When powering via 18650 one of the components (highlighted in picture) heats up enough to smoke a little after a few seconds. Tried with a range of batteries and still get the same. 
It still works fully via USB power with no battery installed, running the last program I flashed to it.

[image]

lewisxhe, 2020-10-28:
T22_V06 There is no battery reverse connection protection. If the battery is reversed incorrectly, it may cause damage @peterwishart

peterwishart, 2020-10-29:
_As I recall_ it has always had the battery installed with correct polarity, and worked before being stored.
It was left with the battery installed, which was a bad idea.

closed 2020-12-04 by lewisxhe