Xinyuan-LilyGO/LilyGo-LoRa-Series #14: Charging LED problem

SeungRyeol, 2020-04-16:
Hi. I received 2 T-beam T22_v1.0 today.

If the charging cable is plugged in while the battery is attached, one of the blue LEDs will turn on, while the other will not turn on.

The current status is:
DCDC1: ENABLE
DCDC2: ENABLE
LDO2: ENABLE
LDO3: ENABLE
DCDC3: ENABLE
Exten: ENABLE

How can I solve this problem?

lewisxhe, 2020-04-17:
The blue light is the charging indicator, you can use axp.setChgLEDMode (AXP20X_LED_OFF), turn it off, it only lights up when charging the battery, the other is the GPS signal indicator, it flashes when the GPS gets valid data

SeungRyeol, 2020-04-17:
Ok, I understand.

Thanks for your comment.

closed 2020-04-17 by SeungRyeol

dunk8888, 2025-05-19:
Hi i have t beam v1. 2 there are 3 leds one for gps wich is on its own, the other two leds are just under the oled, one is status wich is red i think other is blue, they are flashing together when i charge or not they still flash together ime using latest Meshtastic firmware this only started happening when i flashed that, befor was some other firmware wich maybe factory firmware and leds worked as they ahould, i dont think the battery is even charging the icon says it is but the current when i mesure is about 70mah flickering slightly up and down battery eventually goes dead wont charge is this problem with Meshtastic firmware so you think? Thanks

dunk8888, 2025-05-19:
Also should the board work without battery say just usb? mine doesn't it just turns on and off. Thanks

lewisxhe, 2025-05-20:
You can try to flash our test firmware to test whether it can be charged. PMU can be turned on and off by software, including the blue LED light can also be controlled manually

dunk8888, 2025-05-20:
Ok thanks good idea ile play with pmu i think its conflict with firmware Meshtastic ile have a play with it thanks.

dunk8888, 2025-05-21:
Ive sorted it, i uped the charge current to 1000 changed some other parameters and its fine now cheers

dunk8888, 2025-05-21:
Ive just disabled the leds there is 3 on mine, the battery led never comes on whatever the charge status, its flasing in sink with the other, my t beam is v1. 2 axp2101