Xinyuan-LilyGO/LilyGo-LoRa-Series #15: No 3.3v power when running on battery?

nebukadnezar, 2020-04-22:
I'm trying to power a SSD1306 display from the 3.3v supply. Power management is started as follows:

`  axp.setPowerOutPut(AXP192_LDO2, AXP202_ON);
  axp.setPowerOutPut(AXP192_LDO3, AXP202_ON);
  axp.setPowerOutPut(AXP192_DCDC2, AXP202_ON);
  axp.setPowerOutPut(AXP192_EXTEN, AXP202_ON);
  axp.setPowerOutPut(AXP192_DCDC1, AXP202_ON);
  
  axp.setDCDC1Voltage(3300);`

This correctly sets the 3.3v supply (which was 1.8v before using setDCDC1(3300)) - but only when USB is connected.

How do I enable this while running on battery?

lewisxhe, 2020-04-23:
It only needs to be set once. This setting is regardless of whether it is battery-powered or USB-powered.

nebukadnezar, 2020-04-23:
Yes, I'm only setting the power settings in the startup. But can you tell me which parameter I need to set so the 3.3v points has power when on battery? All 3.3v power points show 0.0v when running on battery, and only when USB is plugged in do they show 3.3v.

lewisxhe, 2020-04-23:
You only need to set the following two lines, it can turn on the power
axp.setPowerOutPut (AXP192_DCDC1, AXP202_ON);
axp.setDCDC1Voltage (3300);

nebukadnezar, 2020-04-24:
Well, that was user error on my part! After inserting the battery, it helps to push the power button to turn the TTGO on... Can be closed as fixed. The axp functions work as described.

closed 2020-12-04 by lewisxhe