Xinyuan-LilyGO/LilyGo-LoRa-Series #18: Lora conf - not receiving

xefiji, 2020-06-07:
Hi, 

is it possible to send a lora payload from a T-Beam and receive it from a LoRa32 SX1276 OLED (v2) ? I if follow those two tutorials:
https://randomnerdtutorials.com/ttgo-lora32-sx1276-arduino-ide/ or https://randomnerdtutorials.com/esp32-lora-rfm95-transceiver-arduino-ide/, i can send and receive without problem from two LoRa32 devices.

But if i let the LoRa32 receiver ON and tries to send from a T-Beam T22V1.0, nothing seems to happen.

T-beam code: (some gps related vars are set but i removed gps code for clarity. GPS does work fine.)

What am i doing wrong ?

(if i am not in the good place for that kind of beginner issue, please tell me and i'll try to find a more suitable place for this..)

```
#include <TinyGPS++.h>
#include <axp20x.h>
#include <SPI.h>
#include <LoRa.h>

#define SCK     5    // GPIO5  -- SX1278's SCK
#define MISO    19   // GPIO19 -- SX1278's MISnO
#define MOSI    27   // GPIO27 -- SX1278's MOSI
#define SS      18   // GPIO18 -- SX1278's CS
#define RST     23   // GPIO14 -- SX1278's RESET
#define DI0     26   // GPIO26 -- SX1278's IRQ(Interrupt Request)
#define BAND  868E6

TinyGPSPlus gps;
HardwareSerial GPS(1);
AXP20X_Class axp;

void setup()
{
  Serial.begin(115200);
  Wire.begin(21, 22);
  if (!axp.begin(Wire, AXP192_SLAVE_ADDRESS)) {
    Serial.println("AXP192 Begin PASS");
  } else {
    Serial.println("AXP192 Begin FAIL");
  }
  axp.setPowerOutPut(AXP192_LDO2, AXP202_ON); //lora
  axp.setPowerOutPut(AXP192_LDO3, AXP202_ON); //gps
  axp.setPowerOutPut(AXP192_DCDC2, AXP202_ON); 
  axp.setPowerOutPut(AXP192_EXTEN, AXP202_ON);
  axp.setPowerOutPut(AXP192_DCDC1, AXP202_ON); //oled
  GPS.begin(9600, SERIAL_8N1, 34, 12);   //17-TX 18-RX

  SPI.begin(SCK,MISO,MOSI,SS);
  LoRa.setPins(SS,RST,DI0);
  if (!LoRa.begin(BAND)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }else{
    Serial.println("Starting LoRa succeed!");
  }
  
  axp.setChgLEDMode(AXP20X_LED_BLINK_1HZ);

}

void loop()
{
  LoRa.beginPacket();
  LoRa.print("Hello");  
  LoRa.endPacket();
  Serial.println("sent..");

  delay(1000);
}
```
Thanks :)

lewisxhe, 2020-06-08:
I noticed that you only turned on the LD2 power, but did not set its operating voltage, you can try to add
```
axp.setLDO2Voltage(3300); //LORA VDD set 3v3,
```
 but I think this is not the main reason. You can try it first.

xefiji, 2020-06-09:
Hi @lewisxhe , 

i tried to change it and no change.
I tried to use https://github.com/Extentsoftware/TBeamPower.

Still nothing.
Here is the code:

```

#include <Arduino.h>
#include <SPI.h>
#include <LoRa.h>         // https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md
#include "gps.h"
#include "TBeamPower.h"

TBeamPower power(PWRSDA, PWRSCL); //21 22

void setupSerial() { 
  Serial.println();
  power.print_wakeup_reason();
}

void setup() {

  Serial.begin(115200);
  while (!Serial);

  Serial.println("Power begin");
  power.begin();

  Serial.println("All sensors off");
  power.power_sensors(false);

  Serial.println("All peripherals off");
  power.power_peripherals(false);

  Serial.println("Powering LoRa");
  power.power_LoRa(true);

  SPI.begin(SCK,MISO,MOSI,SS); //5 19 27 18
  LoRa.setPins(SS,RST,DI0); //18 14 26

  if (!LoRa.begin(BAND)) { //868E6 (France)
    Serial.println("Starting LoRa failed!");
    while (1);
  }else{
    Serial.println("Starting LoRa succeed!");
  }

  Serial.println();
}

void loop() {
  power.led_onoff(true);
  delay(500);
  power.led_onoff(false);

  LoRa.beginPacket();
  LoRa.print("hello");  
  LoRa.endPacket();
  Serial.println("sent..");
  power.print_status();
  Serial.println();
  delay(2000);  
}
```
And here is the output i get

```
sent..
Voltages:
         DCDC1: 3.30v
         DCDC2: 1.25v
         DCDC3: 3.30v
         LDO2: 3.30v
         LDO3: 2.80v
ChargeCurrent:   0.70A
IPSOUTVoltage:   3.57v
Temp:            15.94°C
TSTemp:          800.00
VbusCurrent:     0.00
VbusVoltage:     0.00
Battery:
         Connected: false
         Charging:  false
         ChargEN :  true
         Voltage:   0.00v
         Inpower:   0.00
         DischgCur: 0.00
         ChargeCur: 0.00
```

I even tried to change the antenna.

bmcfeeters, 2020-10-21:
I had a similar issue and it turned out that the jgromes/RadioLib library that the T-Beam was using has a default LORA spread factor of 9 and the sandeepmistry/arduino-LoRa library I was using on the LoRa32 SX1276 OLED and the default SF there was 7.  Once I set them to the same value I was able to send and receive LORA packets between them.

Mohamedx7, 2020-10-21:
> I had a similar issue and it turned out that the jgromes/RadioLib library that the T-Beam was using has a default LORA spread factor of 9 and the sandeepmistry/arduino-LoRa library I was using on [...]

from where you can change these settings?

mosfeets, 2020-10-21:
With the RadioLib library the spread factor is set to the default value of 9 in the begin() method for the individual SX1276 module [SX1276.h](https://github.com/jgromes/RadioLib/blob/master/src/modules/SX127x/SX1276.h) file.  It can be changed by passing in a custom value for that parameter to begin() or by using the setSpreadingFactor() method.

For the arduino-LoRa library, there is just the setSpreadingFactor() method as defined in [LoRa.h](https://github.com/sandeepmistry/arduino-LoRa/blob/master/src/LoRa.h)

closed 2020-12-04 by lewisxhe