Xinyuan-LilyGO/LilyGo-LoRa-Series #20: GPS problems...

CristalLagoons-BArratia, 2020-06-22:
Hi! I have two t-beam with the AXP192 chip and I'm struggling to achieve GPS information. The OLED is working and the LoRa functionality too, I can send and receive messages, but when trying to get any GPS data it keeps saying "Positioning" and then change to "NO GPS detected". 
I tried disconnecting the OLED display and checking the serial monitor, here is what it says:

> 
[image]

Please help! 

Thanks in advance :)

CristalLagoons-BArratia, 2020-06-22:
Solved, It was a I2C problem...

closed 2020-06-22 by CristalLagoons-BArratia

Zapfmeister, 2020-08-19:
What exactly was the issue?

CristalLagoons-BArratia, 2020-08-26:
It was a while ago, but I think that it was a problem with the TX and RX pins, I switched them and It worked (I think I was using a bad or old image for reference)

CristalLagoons-BArratia, 2020-08-26:
> What exactly was the issue?

It was a while ago, but I think that it was a problem with the TX and RX pins, I switched them and It worked (I think I was using a bad or old image for reference)