Xinyuan-LilyGO/LilyGo-LoRa-Series #25: Which pin to use for interrupt on t-beam?

geronimoooooooo, 2020-08-25:
Hey,

I have here several different ttgo t-beams. 
a) TT_V07 with 20180711
b) TT_V1.0 with 20190612

Those 2 versions have different pins on the board.

And I use a 10DOF GY MPU9250 external sensor. This sensor has a pin for INTERRUPT. So every time you move the sensor it sends a signal on the INT pin.
When I use the external MPU9250 with TT_V07 and wire the INT from the MPU to the pin 35 on the t-beam, I can read the signal. It switches from LOW to HIGH. You can use this to wake up from deep sleep, because the pin 35 is RTC.

But when I use the same sensor with the TT_V1.0 on the same pin 35, there is no signal. I already tested it with 2 different TT_V1.0.

Which pin do I use on the TT_V1.0 to get an signal/interrupt from an external sensor? I just want to read it when it switches LOW/HIGH.

I am kinda stuck, because I would like to use motion from the mpu to wake up the esp32 from deep sleep. And I need this interrupt  to wake up(that works on V07, but does not work on V1.0).

Any help will be greatly appreciated

lewisxhe, 2020-08-25:
V1.0 35 Pin seems to have been used for axp192 power interruption, you can replace it with another pin

geronimoooooooo, 2020-08-25:
yes, i figured out that 35 is used else where, but 33 or 32 didn't work either. 34 is used for communication

Can you suggest any other pin?

lewisxhe, 2020-08-25:
32, 33 have been connected to dio0 and dio1 of lora. You can replace other ones, such as 25, 26. I don’t remember which ones are not used, you can check the schematics in github

lewisxhe, 2020-08-25:
https://github.com/Xinyuan-LilyGO/LilyGO-T-Beam/blob/master/schematic/LilyGo_TBeam_V1.0.pdf

geronimoooooooo, 2020-08-25:
Ok, I will make a fast test on pin 25, 26 and report here. Let's hope it works :)

geronimoooooooo, 2020-08-25:
awesome! pin25 seems to do the job.

big thx lewisxhe!

lewisxhe, 2020-08-25:
Cheers

closed 2020-08-25 by lewisxhe

eupanicroom, 2023-05-11:
@geronimoooooooo  what i should add to have working wake on motion with the T-beam v1.1 mpu6050 and this git: https://github.com/Max-Plastix/tbeam-helium-mapper

Thank you for reply