Xinyuan-LilyGO/LilyGo-LoRa-Series #26: Error after flash

Waldmensch1, 2020-09-02:
I checked out build and uploaded the sketch with platformIO. I only changed the Radio frequency to 868 in board_def.h. The rest is unchanged. Is the Radio chip selection wrong? How I find out which chip is on the board?

```
--- Miniterm on COM10  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
I2C device found at address 0x34 !
axp192 PMU found
I2C device found at address 0x3C !
ssd1306 display found
done

AXP192 Begin PASS
DCDC1: ENABLE
DCDC2: ENABLE
LDO2: ENABLE
LDO3: ENABLE
DCDC3: ENABLE
Exten: ENABLE
----------------------------------------
DCDC1: ENABLE
DCDC2: ENABLE
LDO2: ENABLE
LDO3: ENABLE
DCDC3: ENABLE
Exten: ENABLE
SSD1306 Begin PASS
[Radio] Initializing ... failed, code -705

```

pbgben, 2020-09-09:
I had same problem, its because the RADIO_TYPE setting in board def, I have EU 868 too and had to change my radio to SX1276 - See below

/*
Replace the model according to the actual situation

RADIO_TYPE option:
            - SX1278
            - SX1276
            - SX1262
*/
#define RADIO_TYPE      SX1276

arneschlag, 2020-10-03:
> I had same problem, its because the RADIO_TYPE setting in board def, I have EU 868 too and had to change my radio to SX1276 - See below
> 
> /*
> Replace the model according to the actual situation
> 
> RADIO_TYPE option:
> - SX1278
> - SX1276
> - SX1262
> */
> #define RADIO_TYPE SX1276

For me this fixed the issue

lewisxhe, 2021-01-19:
The sample code has been updated. The problem has not been updated for a long time and will be closed

closed 2021-01-19 by lewisxhe