Xinyuan-LilyGO/LilyGo-LoRa-Series #27: Feature request: Simple example how to build a simple Battery Voltage check

Waldmensch1, 2020-09-02:
The Example Source is very mightful. Is there a simple approach to have a simple Battery Voltage check? In earlier Board Revisions only GPIO 35 was needed to analog read because there was a Voltage divider on vs. Battery power. I tried to move the code from example partially over to an existing project but axp.getBattVoltage() returns always 3.00 to me 

I use in Setup

```
  axp.begin(Wire, AXP192_SLAVE_ADDRESS);

  Serial.printf("DCDC1: %s\n", axp.isDCDC1Enable() ? "ENABLE" : "DISABLE");
  Serial.printf("DCDC2: %s\n", axp.isDCDC2Enable() ? "ENABLE" : "DISABLE");
  Serial.printf("LDO2: %s\n", axp.isLDO2Enable() ? "ENABLE" : "DISABLE");
  Serial.printf("LDO3: %s\n", axp.isLDO3Enable() ? "ENABLE" : "DISABLE");
  Serial.printf("DCDC3: %s\n", axp.isDCDC3Enable() ? "ENABLE" : "DISABLE");
  Serial.printf("Exten: %s\n", axp.isExtenEnable() ? "ENABLE" : "DISABLE");

  Serial.println("----------------------------------------");

  pinMode(PMU_IRQ, INPUT_PULLUP);
  attachInterrupt(PMU_IRQ, [] {
      pmu_irq = true;
  }, FALLING);

  axp.adc1Enable(AXP202_BATT_CUR_ADC1, 1);
  axp.enableIRQ(AXP202_VBUS_REMOVED_IRQ | AXP202_VBUS_CONNECT_IRQ | AXP202_BATT_REMOVED_IRQ | AXP202_BATT_CONNECT_IRQ, 1);
  axp.clearIRQ();

  if (axp.isChargeing()) {
      baChStatus = "Charging";
      Serial.println(baChStatus);
      Serial.printf("Voltage: %f\n", axp.getBattVoltage());
  }
```
and in loop. I don't understand why pmu_irq is set to false and never again to true. So this is only executed once. Do I have to set all the flags either how it's done in the example?

```
  if (axp192_found && pmu_irq) {
        pmu_irq = false;
        axp.readIRQ();
        if (axp.isChargingIRQ()) {
            baChStatus = "Charging";
        } else {
            baChStatus = "No Charging";
        }
        if (axp.isVbusRemoveIRQ()) {
            baChStatus = "No Charging";
        }
        digitalWrite(2, !digitalRead(2));
        axp.clearIRQ();
    }

  vbat = axp.getBattVoltage();
```

lewisxhe, 2020-09-12:
Yes, the flag should be reset. This is a problem. Thank you for your reminder. This is an early sample program, which will be changed later and may be ignored.

lewisxhe, 2020-09-12:
Regarding axp192, you can check here, I think you only enable the adc current detection, but not the voltage function, please check here
https://github.com/lewisxhe/AXP202X_Library/blob/master/examples/axp_adc/axp_adc.ino

closed 2020-09-12 by lewisxhe