Xinyuan-LilyGO/LilyGo-LoRa-Series #28: Cant get a GPS Fix

arneschlag, 2020-10-03:
I have Problems getting a valid GPS Location, I think there is a Problem in the communication between the gps chip and esp.
I am using the T-Beam Board with V1.1, and when running the script the red led next to the GPS module is blinking. 
But i cant receive any data over Serial1.

lewisxhe, 2020-10-04:
Have you tested it?  Can ESP32 and GPS module communicate?

arneschlag, 2020-10-04:
> Have you tested it? Can ESP32 and GPS module communicate?
Do you have a Idea how could I test it?

lewisxhe, 2020-10-06:
Try the following test code, if normal, then the serial port should output GPSinformation

```
#include <Arduino.h>

#define SerialMon Serial1

// V1.0 version
#define PIN_TX      12
#define PIN_RX      34

bool reply = false;

void setup()
{
    Serial.begin(9600);
    SerialMon.begin(9600, SERIAL_8N1, PIN_RX, PIN_TX);
}

void loop()
{
    if (SerialMon.available()) {
        Serial.write(SerialMon.read());
    }
}

```

lewisxhe, 2020-10-08:
Please use the example here to test it again, under normal circumstances, it will output GPS information on the serial port. If the hardware is damaged, no information will be output

https://github.com/Xinyuan-LilyGO/LilyGO-T-Beam/tree/master/examples/gps

arneschlag, 2020-10-09:
ets Jun  8 2016 00:22:57
I tested it, with this script, but only Information is:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

No GPS Data, so the Hardware is damaged?

lewisxhe, 2020-10-10:
Unfortunately, the communication between other GPS and esp32 is not normal, if you just bought it, you can contact LilyGo sales

kd8bxp, 2020-10-16:
I had a similar problem, the red led would blink, but I was getting no data output. It turns out, the NMEA stream from the device was turned off (how this happened, or why it happened I have no idea). The only library that I found that was able to fix it was the Sparkfun Ublox Arduino library https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library Example 6 and Example 12
I was never able to get the Sparkfun library to give me a location however.  But the tinygps++ library was able to display my location, time, how many sats, ect.  So between the two I was able to get my GPS units to start to work again.
Hope that helps someone out.

lyusupov, 2020-10-17:
**FYI**:
Issue report: https://github.com/meshtastic/Meshtastic-device/issues/334
Workaround: https://github.com/eriktheV-king/TTGO_T-beam_GPS-reset

lewisxhe, 2020-10-28:
@arneschlag  Have you tried @lyusupov's method? If it is resolved, please let me know.

arneschlag, 2020-10-29:
> @arneschlag Have you tried @lyusupov's method? If it is resolved, please let me know.

I tried @lyusupov method and it worked (kind of), i get some data from the gps chip, but until now no coordinates. 
In the next following days I will try to get it working, and will updating then this issue.

FazerTom, 2020-11-08:
I have made the experience that the original GPS antenna (15mm x 5mm) is very weak in reception. I got a fix only after a very long time with absolutely free view to the sky. So I exchanged the antenna for another one (27mm x 27mm x 6.5mm). Now I also get a fix in the room in shortest time.

closed 2021-01-19 by lewisxhe