Xinyuan-LilyGO/LilyGo-LoRa-Series #29: Upload sketch error

pbravoz2, 2020-10-08:
Hi there, can anybody help me for this issue??:  uploading the sketch to T-Beam throws me the classic ESP32 error: **Failed to connect to ESP32: Timed out waiting for packet header**. The solution for the other ESP32 boards is to place a 10uF capacitor between the EN PIN and GND. But **T-Beam doesn't have the EN pin!** How can I solve it?

I am very thankfull for your help.
Greetings, Pedro

lewisxhe, 2020-10-12:
You can try it like this,
IO0 is shorted to GND, then press reset, then click upload

pbravoz2, 2020-10-13:
Hi lewisxhe, it works perfect. The order is:

-IO0 shorted to GND
-reset (it boots in a special mode waiting for upload)
-upload the sketch
-disconnect IO0 from GND
-reset to normal boot

Thank you!!

closed 2020-10-13 by pbravoz2