Xinyuan-LilyGO/LilyGo-LoRa-Series #33: No packets from another T-Beam can be received.

FazerTom, 2020-11-08:
There may be an error in the Receive Mode function. No packets from another T-Beam can be received. 
The function "radio.startReceive()" uses DIO0 as interup output.  (Quote "Interrupt-driven receive method. DIO0 will be activated when full valid packet is received")

After the following change in the following lines the reception of LoRa packets works.

Line 45:
original: 
  RADIO_TYPE radio = new Module(LORA_SS, LORA_DIO1, LORA_RST, LORA_BUSY);
change:
  RADIO_TYPE radio = new Module(LORA_SS, LORA_DI0, LORA_RST, LORA_BUSY);

line 271:
original: 
  radio.setDio1Action(setFlag);
change:
  radio.setDio0Action(setFlag);

Translated with www.DeepL.com/Translator (free version)

axpaul, 2020-11-14:
Thank, yes the interrup functiannate now with your correction. Thank !

lewisxhe, 2021-01-19:
The problem has been inactive for a long time and will be closed

closed 2021-01-19 by lewisxhe