Xinyuan-LilyGO/LilyGo-LoRa-Series #37: Compile error - Allfunction.ino

smarti2019, 2020-11-24:
AllFunction:84:30: error: no matching function for call to 'Button2::Button2()'

     pBtns = new Button2 [args];

phfu, 2020-12-06:
Hello
I replaced the code with : 
pBtns = new Button2 (BUTTONS_MAP);
and it worked.   It's a bit confusing just for handling one button.

Anyhow, it is then possible to switch betweeb the 3 screens.

GPS works,  LoRa transmission also (I can see the packet on an spectrum analyser) but receiver is not working.
With 2 modules, one TX, other RX,   RX not working,  even when switching the 2 devices.

Regards
Philippe

lewisxhe, 2020-12-08:
The sample code has been updated, please update, it should be easier to understand than the original

lewisxhe, 2021-01-19:
Issues have not been active for a long time and will be closed

closed 2021-01-19 by lewisxhe