Xinyuan-LilyGO/LilyGo-LoRa-Series #38: "ArduinoLoRa" example not work with "T-Beam V1.1 SX1262 LORA"

izzet-kalinsazlioglu, 2020-12-03:
T Beam "ArduinoLoRa" this example use <LoRa.h> library .  
but  this library 
https://github.com/sandeepmistry/arduino-LoRa
not support yet SX1262 LORA chips. 

message is : Starting LoRa failed!

lyusupov, 2020-12-03:
This is not true.

T-Beam [SX1262-specific examples](https://github.com/Xinyuan-LilyGO/LilyGO-T-Beam/tree/master/examples/SX1262) are using [RadioLib](https://github.com/jgromes/RadioLib) instead:
https://github.com/Xinyuan-LilyGO/LilyGO-T-Beam/blob/master/examples/SX1262/SX126x_Receive_Interrupt/SX126x_Receive_Interrupt.ino#L20

RadioLib has declared support for SX162 radio:
https://github.com/jgromes/RadioLib#supported-modules

izzet-kalinsazlioglu, 2020-12-07:
@lyusupov thank you. it works.  I solved with some warnings. 
current problem is . I need Lora module pinouts on this page 
https://www.aliexpress.com/item/4000542688096.html?spm=a2g0o.cart.0.0.6eee3c00vKLQsQ&mp=1
because pin numbers are not explained.
#define LORA_DI0        ??
#define LORA_DIO1       ??
#define LORA_BUSY       ??

i try . it works with some warnings.
#define LORA_DI0        33
#define LORA_DIO1       32
#define LORA_BUSY       34

or 
How can i fix this lora library for sx1262 support. 
https://github.com/sandeepmistry/arduino-LoRa

[image]

[image]

lewisxhe, 2020-12-08:
arduino-LoRa does not support 1262, you can use radiolib

dragosmacovei, 2020-12-26:
> Your statement is not correct.
> 
> T-Beam [SX1262-specific examples](https://github.com/Xinyuan-LilyGO/LilyGO-T-Beam/tree/master/examples/SX1262) are using [RadioLib](https://github.com/jgromes/RadioLib) instead:
> https://github.com/Xinyuan-LilyGO/LilyGO-T-Beam/blob/master/examples/SX1262/SX126x_Receive_Interrupt/SX126x_Receive_Interrupt.ino#L20
> 
> RadioLib has declared support for SX1262 radio:
> https://github.com/jgromes/RadioLib#supported-modules
> 
> So you need to pre-install the RadioLib in order to use SX1262 specific examples with this particular T-Beam variant.
> A copy of the RadioLib is already located in [libdeps](https://github.com/Xinyuan-LilyGO/LilyGO-T-Beam/tree/master/libdeps) folder of this GitHub repo.

The link for git is not working anymore

lewisxhe, 2021-01-19:
Issues have not been active for a long time and will be closed

closed 2021-01-19 by lewisxhe