Xinyuan-LilyGO/LilyGo-LoRa-Series #39: Is it possible to power use the esp_sleep_enable_timer_wakeup

pimdestoere, 2020-12-03:
Hey there, 
Does anybody know a method for using a timer to wake up the t-beam v1.1 from sleep. So without using the button? Or is this not possible with the power management unit?

Some example code would be much appreciated. 

Thanks.

lewisxhe, 2020-12-04:
like this .
```
#define uS_TO_S_FACTOR 1000000ULL  /* Conversion factor for micro seconds to seconds */
#define TIME_TO_SLEEP  15          /* Time ESP32 will go to sleep (in seconds) */

    esp_sleep_enable_timer_wakeup(uS_TO_S_FACTOR * TIME_TO_SLEEP);
    esp_deep_sleep_start();
```

You use the esp32 timer to wake up, you cannot use power management to wake up, it does not support

lewisxhe, 2020-12-04:
I suddenly have an idea, I think it should be possible! I will try to give feedback, if possible I will upload the sample code

pimdestoere, 2020-12-04:
@lewisxhe that's a real shame, i was really hoping for uA efficiency with the t-beam and esp32 sleeptime. 

I have however, found a sort of workaround, the gps system draws the most power, so if you switch that off, either by software of by removing it from the board entirely you get instead 5 mA instead of 70 mA powerdraw using  esp_sleep_enable_timer_wakeup.  Not the uA range i was looking for but an improvement nonetheless. 

If anybody finds a way to get into the uA range anyways, lets me know.

lewisxhe, 2020-12-04:
Do you mean to set all modules on the board to sleep and have 5mA current consumption?

pimdestoere, 2020-12-04:
No i just used the standard esp32 deepsleepmode without using the axp192. And with a removed gps module.

lewisxhe, 2020-12-04:
Are you using USB power supply in the measurement? If it is, then the current consumption is CP2104, please use the battery test. It will be accurate. I have tested that the sleep is about 600uA

pimdestoere, 2020-12-04:
I am measuring using a battery power supply. using:

```
#define uS_TO_S_FACTOR 1000000ULL  /* Conversion factor for micro seconds to seconds */
#define TIME_TO_SLEEP  15          /* Time ESP32 will go to sleep (in seconds) */
    esp_sleep_enable_timer_wakeup(uS_TO_S_FACTOR * TIME_TO_SLEEP);
    esp_deep_sleep_start();
```

With the gps module still attached i  get around 70 mA with it removed around 5mA. 

I do get around 6 uA when using the axp192 method but as stated previously that method does not actually have a way for using the timer to wake up the device.

lewisxhe, 2020-12-04:
Can you paste the code, I don’t quite understand what you mean, the above code will set esp32 to deep sleep

cyberman54, 2020-12-07:
With a T-BEAM v1.0, AXP192 set to sleep mode, LORA and GPS power cut off by AXP192, and a SSD1306 OLED connected to I2C pins and in sleep mode (because cutting off power of OLED would block i2c bus) i see 

- 1,2 mA battery current, if esp wakeup is set by GPIO
- 0,9 mA battery current, if esp wakeup is set by timer (as in example above)

lewisxhe, 2020-12-08:
To use GPIO to wake up, you can use esp_sleep_enable_ext1_wakeup, which can be reduced to about 600uA instead of esp_sleep_enable_ext0_wakeup!

When the timer wakes up, its sleep current is slightly higher than esp_sleep_enable_ext1_wakeup

@cyberman54

lewisxhe, 2020-12-08:
If you use GPIO to wake up, you can use the axp sleep function, it can be lowered, refer to https://github.com/lewisxhe/AXP202X_Library/tree/master/examples/axp_sleep_mode

cyberman54, 2020-12-18:
@lewisxhe But... on T-Beam 1.0/1.1 no GPIO pin of AXP192 is connected to ESP32 MCU, thus it is not possible to use sleep mode of AX192 and wake it up by MCU. Right?

lewisxhe, 2020-12-19:
Even if it is led out, the wake-up function cannot be achieved.  This example demonstrates how to set sleep through AXP202, which can only be awakened by PEK button of AXP202.

lyusupov, 2020-12-24:
Lewis, thank you for the AXP20X library and _axp_sleep_mode_ example

Below is a 'gist' with an example of three ESP32+AXP192 sleep/off mode(s) usage:

```
//#define PMK2_SLEEP_MODE 1    // 0.6 mA : esp_deep_sleep_start()
//#define PMK2_SLEEP_MODE 2    // 0.9 mA : axp.setSleep()
#define PMK2_SLEEP_MODE 3      //  60 uA : axp.shutdown()

    axp.setChgLEDMode(AXP20X_LED_OFF);

#if PMK2_SLEEP_MODE == 2
    int ret;
    // PEK or GPIO edge wake-up function enable setting in Sleep mode
    do {
        // In order to ensure that it is set correctly,
        // the loop waits for it to return the correct return value
        ret = axp.setSleep();
        delay(500);
    } while (ret != AXP_PASS) ;

    // Turn off all power channels, only use PEK or AXP GPIO to wake up

    // After setting AXP202/AXP192 to sleep,
    // it will start to record the status of the power channel that was turned off after setting,
    // it will restore the previously set state after PEK button or GPIO wake up

#endif /* PMK2_SLEEP_MODE */

    axp.setPowerOutPut(AXP192_LDO2, AXP202_OFF);
    axp.setPowerOutPut(AXP192_LDO3, AXP202_OFF);
    axp.setPowerOutPut(AXP192_DCDC2, AXP202_OFF);

    /* workaround against AXP I2C access blocking by 'noname' OLED */
    axp.setPowerOutPut(AXP192_DCDC1, AXP202_OFF);
    axp.setPowerOutPut(AXP192_EXTEN, AXP202_OFF);

    delay(20);

    /*
     * When driven by SoftRF firmware the V08+ T-Beam takes:
     * in 'full power' - 160 - 180 mA
     * in 'stand by'   - 600 - 900 uA
     * in 'power off'  -  50 -  90 uA
     * of current from 3.7V battery
     */
#if   PMK2_SLEEP_MODE == 1
    /* Deep sleep with wakeup by power button click */
    esp_sleep_enable_ext1_wakeup(1ULL << SOC_GPIO_PIN_TBEAM_V08_PMU_IRQ,
                                 ESP_EXT1_WAKEUP_ALL_LOW);
    esp_deep_sleep_start();
#elif PMK2_SLEEP_MODE == 2
    // Cut MCU power off, PMU remains in sleep until wakeup by PEK button press
    axp.setPowerOutPut(AXP192_DCDC3, AXP202_OFF);
#else
    /*
     * Complete power off
     *
     * to power back on either:
     * - press and hold PWR button for 1-2 seconds then release, or
     * - cycle micro-USB power
     */
    axp.shutdown();
#endif /* PMK2_SLEEP_MODE */
```

cyberman54, 2020-12-24:
@lyusupov any chance to bypass this i2c OLED deadlock issue, e.g. by setting GPIO pullups of ESP32 before sleep?

```
/* workaround against AXP I2C access blocking by 'noname' OLED */
    //axp.setPowerOutPut(AXP192_DCDC1, AXP202_OFF);
```

lyusupov, 2020-12-24:
@cyberman54 
What particular GPIO pullups ?
I<sup>2</sup>C pullups are always ON as long as DCDC1 and/or DCDC3 are ON:

[image]

lyusupov, 2020-12-24:
the picture above is  for V0.8-1.0

V1.1 is different there:

[image]

cyberman54, 2020-12-25:
@lyusupov since we have 10Kohm hardwired pullups on T-Beam boards tieing SDA (IO21) and SCL (IO22) to +3V3, idea is to switch on internal pullups of ESP32 in GPIO21/GPIO22, to get a voltage divider. Maybe this can help to unblock i2c bus while a connected OLED display is powered off by cutting off DCDC1.

This would save ~10 - 20 µA power consumption of OLED display/SSD1306 controller, but comes with the price of power consumption of the voltage divider.

Theory so far, i did not test it.

lewisxhe, 2021-01-19:
Issues have not been active for a long time and will be closed

closed 2021-01-19 by lewisxhe