Xinyuan-LilyGO/LilyGo-LoRa-Series #40: 3.3V line sitting at 1.53V

GregorR1, 2020-12-04:
I was using two T-beams with the display modules, and everything was working as expected.  However, a couple of days ago one of my displays would not turn on.  After investigating, I have found that the 3.3V line on this board is sitting at around 1.5V.  The ESP appears to still be running, but the display won't turn on due to the low voltage.

I have tried using the code found at https://github.com/LilyGO/TTGO-T-Beam/issues/22

But this did not make a difference.  

Any help would be greatly appreciated as this is my first time using these boards and I have never worked with a Power controller like this before.

Thanks

lewisxhe, 2020-12-07:
Sorry, the 3.3V pin is controlled by AXP192, DC1, you should add the following two lines after initializing axp192 to turn it on

```
PMU.setDCDC1Voltage(3300);  
PMU.setPowerOutPut(AXP192_DCDC1, AXP202_ON);

```

GregorR1, 2020-12-07:
I'll give that a try justnow.  Thanks

GregorR1, 2020-12-07:
Still no luck, 3.3V pin is still at 1.5V.

This is the sketch I uploaded.
`#include <axp20x.h> 

AXP20X_Class PMU;

// the setup function runs once when you press reset or power the board
void setup() {
  pinMode(4, OUTPUT);
  PMU.setDCDC1Voltage(3300);  
  PMU.setPowerOutPut(AXP192_DCDC1, AXP202_ON);
}
`

lewisxhe, 2020-12-08:
You have not initialized the PMU, I have added instructions in the sample code, please refer to here:

https://github.com/Xinyuan-LilyGO/LilyGO-T-Beam/blob/866e164ee08b4c7a0263ceaf801e87afadcf0140/examples/ArduinoLoRa/LoRaSender/boards.h#L55

GregorR1, 2020-12-08:
Hi, I ran the example code LoRaSender that you linked, and the LoRa module is sending correctly.  I can receive the LoRa packets with my other T-Beam. 

However, it hasn't solved the issue as the 3.3V line is still at 1.5V

lewisxhe, 2020-12-09:
You can use PMU.getDCDC1Voltage() to view the returned voltage!

Then you can use
```
PMU.setDCDC1Voltage(1800)
Serial.println(PMU.getDCDC1Voltage());
```

If the set voltage does not match the read voltage, it may be broken.

If the setting is the same as the read voltage, but the actual measured voltage is different, then it is also broken!

You'd better take a picture and show me where the voltage is!

GregorR1, 2020-12-09:
Hi,

I added these lines, but I appear to be getting a random number from the PMU:

[image]

Any ideas?

lewisxhe, 2020-12-10:
Unfortunately, it is damaged.

GregorR1, 2020-12-10:
Okay, It appears to only be the 3.3V line that is affected for now.  I'll keep using it for testing and will keep in mind that I can't use the 3.3V pins.

Thanks for your help.

lewisxhe, 2021-01-19:
Issues have not been active for a long time and will be closed

closed 2021-01-19 by lewisxhe