Xinyuan-LilyGO/LilyGo-LoRa-Series #41: SD Card through SPI bus on TTGO LoRa T-Beam v0.7

CPlagne, 2020-12-11:
Hello,

I am attempting to add a microSD card reader to my T-Beam for data collection, and after reading documentation have come to the conclusion that ESP32 makes available two SPI buses, VSPI and HSPI. On the T-Beam V0.7, VSPI uses pins 5, 18,19 and 27, and are all wired directly to LoRa and unavailable. Likewise, HSPI uses pins 12, 13, 14 and 15, with 12 and 15 in particular (meant to be MISO and CS) are used for the GPS as TX and RX respectively. In all, that seems to mean that no SPI bus is fully available.

I have seen users work around this by using other GPIO pins, mostly with the aim of adding a screen (see issue #9 in the old LilyGo repo), and one user over the internet in particular for an SD card, claiming his code worked - this has not been the case for me, and the SD card doesn't mount. Here is their code below : 

```
#include <SPI.h>
#include <SD.h>

// HSPI
#define HSPI_SCLK 14
#define HSPI_MISO 4
#define HSPI_MOSI 13
#define HSPI_CS 2

SPIClass * hspi = NULL;
String dataString =""; // holds the data to be written to the SD card
File sensorData;

void setup() {

  hspi = new SPIClass(HSPI);
  hspi->begin(HSPI_SCLK, HSPI_MISO, HSPI_MOSI, HSPI_CS); //SCLK, MISO, MOSI, SS
  pinMode(HSPI_CS, OUTPUT); //HSPI SS
  
  Serial.begin(115200);
  
  // see if the card is present and can be initialized:
  if (!SD.begin(HSPI_CS)) {
    Serial.println("Card failed, or not present");
  }

  if (SD.exists("data.csv")) {
    Serial.println("data.csv exists.");
  }
  else {
    Serial.println("data.csv doesn't exist.");
  }

  // open a new file and immediately close it:
  Serial.println("Creating data.csv...");
  sensorData = SD.open("data.csv", FILE_WRITE);
  sensorData.close();

  // Check to see if the file exists:
  if (SD.exists("data.csv")) {
    Serial.println("data.csv exists.");
  }
  else {
    Serial.println("data.csv doesn't exist.");
  }
  while(1);
}

void loop() {

}
```

This code systematically returns "Card failed, or not present" (and failure to create or read the file, obviously). I have made sure that the microSD card reader has a 5V supply like the datasheet recommands, and that the card is formatted in FAT32 and reads and writes files correctly on my computer. Besides creating a software SPI connection, I'm out of ideas.

Thank you in advance

lewisxhe, 2020-12-12:
You didn't assign SPI to SD Library... it should be like this:
if (!SD.begin(HSPI_CS,*hspi)) {
    Serial.println("Card failed, or not present");
  }

closed 2020-12-28 by CPlagne

FLYINGSC0T, 2021-03-02:
Tried the code with the solution but couldn't get it to work for me on a T-Beam v0.7 running a SPI TFT display, LoRa, GPS and a BMP280 barometer. LoRa stopped working!! My search continues for a working solution...

stiliajohny, 2021-11-21:
Hi all 
Is there any update on this ???