Xinyuan-LilyGO/LilyGo-LoRa-Series #42: LoRa transmission through T-Beam V0.7

CPlagne, 2020-12-21:
Hello,

I'm trying to transmit data through the LoRa antenna between my TTGO T-BEAM V0.7 (emitter), to a Heltec Lora32 V2 (receiver). For this, I'm using the LoraSender example code from this repo, with `#define LILYGO_TBeam_V0_7` uncommented in utilities.h, band 868E6 and with all the libraries and header files set up. For the Heltec, I'm using the heltec library found here : https://github.com/HelTecAutomation/Heltec_ESP32 (OLED_LoRa_Receiver code). The LoRa connection seems to begin with no problem, the serial monitor shows the T-Beam sending packets, but the heltec card receives nothing.

I've tested the heltec card with another similar one and it works fine. It is set at baudrate 115200 and 868E6 band like the T-Beam. I am out of ideas as to why it is not receiving. Please help me move forward.

Thank you,
Clément

lewisxhe, 2020-12-31:
I don’t have a Heltec board. Do you use Radio directly or write Heltec directly on ArduinoLoRa? If he uses the same software version, he can definitely communicate

lewisxhe, 2021-01-19:
Issues have not been active for a long time and will be closed

closed 2021-01-19 by lewisxhe