Xinyuan-LilyGO/LilyGo-LoRa-Series #43: FAILURE

dragosmacovei, 2020-12-26:
```
initBoard
LoRa Receiver

C:\Users\USERNAME\OneDrive\Documents\Arduino\libraries\LMIC-Arduino\src\lmic\radio.c:691
```

This is the output I get with TTN_OTTA example on TTGO T-Beam. I did tried to 

```
#define CFG_eu868 1
//#define CFG_us915 1
// This is the SX1272/SX1273 radio, which is also used on the HopeRF
// RFM92 boards.
#define CFG_sx1272_radio 1
// This is the SX1276/SX1277/SX1278/SX1279 radio, which is also used on
// the HopeRF RFM95 boards.
//#define CFG_sx1276_radio 1
```

change from `CFG_sx1272_radio  `to `CFG_sx1276_radio ` but the same ASSERT fails

lewisxhe, 2020-12-31:
If you use arduino ide, you need to rename TTN_OTTA.cpp to TTN_OTTA.ino. I did not report an error on my computer!
[image]

lewisxhe, 2021-01-19:
Issues have not been active for a long time and will be closed

closed 2021-01-19 by lewisxhe