Xinyuan-LilyGO/LilyGo-LoRa-Series #44: How to power it using a pin

Cuningan, 2020-12-27:
Hello.
I am trying to power t-beam v1.1 using 5V pin or 3.3v pin but it does not work.
How can I power the t-beam externally without using the USB??

Cuningan, 2020-12-27:
It seems that axp192 is not powered on when using external 5v pin, labeled "VSYS" in the schematics.
The 3.3V pins also does not power the board on...

lewisxhe, 2020-12-31:
Do you mean to use one T-Beam to power another T-Beam?

If T-Beam removes the USB input, its 5V will not be output. Because it is connected to VBUS

Cuningan, 2020-12-31:
Hello, no I mean powering the T-beam connecting 5V to the 5V pin.
As per your example, I can connect one T-beam using USB and I can connect the 5V out to other T-beam 5V but the other will not be powered.
It will not work.

lyusupov, 2021-01-02:
VCC and VSYS are "**voltage output only**" pads of the pin header on V0.8-V1.1

[image]

However, you may consider to purchase [older variant of the T-Beam](http://s.click.aliexpress.com/e/EKQW70xe).
It is is still remain in production and is more friendly for external power source connection.

[image]

You can supply 5V as well through one of the power switch pins

[image]

Cuningan, 2021-01-02:
Hello, good year.

And why it was removed in new versions??
It is like losing features.

The only "of the shelf" mode of powering it is using USB.

It limits a lot the options when using it.

lyusupov, 2021-01-02:
[Acording to this page](https://github.com/lyusupov/SoftRF/wiki/Prime-Edition-MkII#battery-life), the V1.x is 75% more power efficient than V0.7 one.
For vast majority of users the power efficiency is much more valuable factor rather than external power connection option(s).
For those few who needs external power - the [V0.7 is still available for pacing orders](http://s.click.aliexpress.com/e/EKQW70xe).

 It is worth to mention that anyone can still supply power to V1.x over:
- micro-USB socket/pins or
- pads of 18650 battery holder

lewisxhe, 2021-01-19:
Issues have not been active for a long time and will be closed

closed 2021-01-19 by lewisxhe

bradleyfalzon, 2021-02-03:
Just to give a +1 to this, we'd also like this feature by connecting the AXP192's ACIN pin to a header or even just a pad like the T-Beam v1.1 has with EXTEN and AXP_I0s.

Importantly, I want to be able to provide a small 6v solar panel to this board. Connecting this directly to the AXP192 ACIN pins appears to be a supported method for a 5w (6v ~1 amp) panel and would safely charge the battery as well as provide the relevant charging information via registers.

Connecting a panel directly to the battery pads with a battery connected isn't an option, and providing more than 5.25v to VBUS is outside of the CP2104's input voltage range.

So in the next iteration, providing access to the AXP192 ACIN pin would be fantastic.

IZ1IVA, 2021-02-18:
I second that. I'd love to power the board from a LiFePO4 cell, even if that means not being able to charge it in-device.