Xinyuan-LilyGO/LilyGo-LoRa-Series #46: Configure T-Beam NEO-M8N

vmotorkin, 2021-01-25:
Hi. I use LILYGO®TTGO T-Beam NEO-M8N for topographic surveys. I need to receive NMEA messages GGA, GLL, GSA, GSV, RMC, VTG. How can I do it? Can i used u-center or Arduino IDE?
Regards.

lewisxhe, 2021-01-28:
Check out the simplest example of GPS module output, it will output all data to the serial port, and then you can process it as needed. 

https://github.com/Xinyuan-LilyGO/LilyGO-T-Beam/blob/master/examples/GPS/Example1_BasicNMEARead/Example1_BasicNMEARead.ino

closed 2021-01-28 by lewisxhe

vmotorkin, 2021-01-28:
What port is the bluetooth on  LilyGO-T-Beam NEO-M8N?
Thanks!
------
Best regards,
Vyacheslav Motorkin
+7 918 4057326

vmotorkin, 2021-01-28:
What port is the bluetooth on  LilyGO-T-Beam NEO-M8N?

lewisxhe, 2021-01-28:
Bluetooth port?  I don't understand what you are talking about. GPS and ESP32 are connected via serial port.  You can only access through the serial port, or you can write a special program to forward data to Bluetooth.