Xinyuan-LilyGO/LilyGo-LoRa-Series #47: Example1_BasicNMEARead only outputs GPGGA

DCWjeboy, 2021-02-13:
I expected it to output  GLL, GSA, GSV, RMC and VTG as well. is there a way to do this also?

flashspys, 2021-02-26:
Can't confirm. I just ran the example and got all mentioned outputs.

LilyGO, 2021-09-11:
It should be a problem with your GPS settings. Each GPS module can individually set what information needs to be output to reduce power consumption. Just check the GPS module of your module, find the communication protocol and check in.

chinswain, 2022-04-10:
You have to enable\disable the message types:

`     myGPS.disableNMEAMessage(UBX_NMEA_GLL, COM_PORT_UART1);
      myGPS.disableNMEAMessage(UBX_NMEA_GSA, COM_PORT_UART1);
      myGPS.disableNMEAMessage(UBX_NMEA_GSV, COM_PORT_UART1);
      myGPS.disableNMEAMessage(UBX_NMEA_VTG, COM_PORT_UART1);
      myGPS.enableNMEAMessage(UBX_NMEA_RMC, COM_PORT_UART1);
      myGPS.disableNMEAMessage(UBX_NMEA_GGA, COM_PORT_UART1);
      myGPS.saveConfiguration(); //Save the current settings to flash and BBR`

Source: https://github.com/eriktheV-king/TTGO_T-beam_GPS-reset/blob/master/T22-GPS-reset-v3/T22-GPS-reset-v3.ino

lewisxhe, 2022-12-08:
There is no activity for a long time. If there is still a problem, please reopen the problem

closed 2022-12-08 by lewisxhe