Xinyuan-LilyGO/LilyGo-LoRa-Series #48: TBeamAllFunction example not working on my board V1.1 !

vincegre, 2021-02-24:
Hi

I'm trying the sample code TBeamAllFunction with my TBeam v1.1 (as labelled on board for version). I have installed all libs and dependencies in Arduino IDE and compiles code properly (I just uncommented the line to indicate I have board 1.1) but once loaded in board nothing happens ! OLED screen stays off and none of button is doing anything (out of rst that briefly switches off the RED solid light).
Any ideas what's wrong there ? or how that code is supposed to work ?

Thanks

Vincèn

vincegre, 2021-02-27:
I tried to change the radio chipset declared in code but out of generating a different error code it doesn't work :(

lewisxhe, 2021-03-01:
Please make sure that the model of the radio frequency chip you use corresponds to that in the program, a wrong choice will result in failure.

vincegre, 2021-03-01:
> Please make sure that the model of the radio frequency chip you use corresponds to that in the program, a wrong choice will result in failure.
As I'm in Europe I use 868 so from what I have read in code I need to do nothing to use it at such frequency (my board is labelled 868/915) and I have also tried to uncomment and force frequency in utilities.h:

`#define LoRa_frequency      868.0`

But no changes it makes an error each time at boot during Radio Initialisation !!  Really bored of that board manufacturer unable to provide support and working sample codes for its boards :-1:

lewisxhe, 2021-03-01:
I do not understand . What do you mean by saying this ?

vincegre, 2021-03-01:
> I do not understand . What do you mean by saying this ?

The manufacturer of these boards offer no support (no one from their company answers here to support their customers) and sample codes they are supplying are not working too....

lewisxhe, 2021-03-01:
Can your board work now?

vincegre, 2021-03-01:
> Can your board work now?

Nope unable to get that demo code to work :-1:

lewisxhe, 2021-03-01:
Then I will compile a binary file tomorrow, please try again, if it still doesn’t work, then don’t hesitate to contact the supplier you bought directly and ask for after-sales

vincegre, 2021-03-01:
> Then I will compile a binary file tomorrow, please try again, if it still doesn’t work, then don’t hesitate to contact the supplier you bought directly and ask for after-sales

Sorry and no offense but it looks like you don't understand at all what I told in my previous message !! Contact with manrufacturer solves absolutely nothing as they speak barely english and don't care so I guess I'll try to sell that board second hand to get rid of it and buy some Lora boards made by a manufacturer that cares about its customers !!

lewisxhe, 2021-03-02:
I added the binary file of allfunction, you can upload and try it, you can send me the link you purchased, and I will check which RF chip it is. 

https://github.com/Xinyuan-LilyGO/LilyGO-T-Beam/tree/master/firmware

vincegre, 2021-03-02:
> I added the binary file of allfunction, you can upload and try it, you can send me the link you purchased, and I will check which RF chip it is.
> 
> https://github.com/Xinyuan-LilyGO/LilyGO-T-Beam/tree/master/firmware

Thanks a lot for adding the binaries and I finally got it working by loading the 1.1 with SX1276 chipset. Did you build these binaries just using the original ino file posted here ? Would you mind to share it so I can check differences between mine not working and your one ??

lewisxhe, 2021-03-02:
I used the allfunction sample compiled in github, please make sure your configuration in "utilities.h" and "TBeamAllFunction.ino" is correct 
[src.zip](https://github.com/Xinyuan-LilyGO/LilyGO-T-Beam/files/6067319/src.zip)

dykst002, 2021-03-03:
Hello lewisxhe. I've been following your discussion so far. I carried out your instructions on my Tbeam today. And it works. I have a display that is hopeful. Unfortunately, I still have a message "No GPS detected", yet the red LED under the GPS chip on the board gently flashes. Also on the serial monitor of arduino-ide comes with the same message No GPS detected. Do you have any solution direction / ideas

lyusupov, 2021-03-08:
https://github.com/Xinyuan-LilyGO/LilyGO-T-Beam/issues/28#issuecomment-710755451

lewisxhe, 2022-12-08:
There is no activity for a long time. If there is still a problem, please reopen the problem

closed 2022-12-08 by lewisxhe