Xinyuan-LilyGO/LilyGo-LoRa-Series #49: Error compiling an 'empty' code on tbeams T22 V1.1

jmarcB, 2021-02-28:
Hi,
First steps here with a t-beam V1.1 under platformio.
When creating a new project from scratch through PIO +New Project , selecting board TTGO T-Beam, thus building it I am getting a strange error :
compilation terminated.
*** [.pio\build\ttgo-t-beam\FrameworkArduino\IPAddress.cpp.o] Error 1
In file included from C:\users\jean-marc\.platformio\packages\framework-arduinoespressif32\cores\esp32\FunctionalInterrupt.cpp:22:0:
C:\users\jean-marc\.platformio\packages\framework-arduinoespressif32\cores\esp32\Arduino.h:185:26: fatal error: pins_arduino.h: No such file or directory

When I look at .plateformio.ini I do see usual parameters
[env:ttgo-t-beam]
platform = espressif32
board = ttgo-t-beam
framework = arduino

When I look at .../variant/ttgo-t-beam  there is a subdirectory containing pins_arduino.h so I am stuck. Probably a path badly setup but I can't figure out where. 

Advise or help is welcome
BR JM

lewisxhe, 2021-03-01:
You should not have completed the installation, like missing variant files, everything is normal here.

```
[env:ttgo-t-beam]
platform = espressif32
board = ttgo-t-beam
framework = arduino
```

closed 2021-03-01 by lewisxhe

jmarcB, 2021-03-01:
HI Lewis, not sure I understood your answer. Unless there is a special installation for this board (where should I find it) , I followed the usual path to create a project within plateformIO. The board is in the list of the board supported so ?.
Happy to get the right informations to attain the right setup as I am starting to make devs on those boards (tbeam & lora32)
Thank you

lewisxhe, 2021-03-01:
There is no special installation method, there is already a variant of T-Beam in platformIO. 

https://github.com/platformio/platform-espressif32/blob/develop/boards/ttgo-t-beam.json

jmarcB, 2021-03-01:
Hi Lewis, furthermore, I think that the issue is that the by default variant is "t-beam" but 
in .platformio\packages\framework-arduinoespressif32\variants the repository name is  "tbeam"
Thus either a "-" is missing in the installed directory or should be avoided  in ttgo-t-beam.json