Xinyuan-LilyGO/LilyGo-LoRa-Series #51: Which SX12## is on my TBEAM v1.1 ?

thypl9, 2021-04-06:
Hello,

I've bought some TTGO TBEAM (without OLED).

On the PCB, I can read : "**T22_V1.1**" and "**20191212**"

The sticker on the GPS says : "Neo-6M-0-001"

On another sticker, there is **only this 3 lines** :
TTGO
868/915Mhz
Modle: LORA32

Nothing about "SX1262", "SX1276" or "SX1278".

When I read the Product part of the README.MD of the main page of the LilyGO-T-Beam project,
here [https://github.com/Xinyuan-LilyGO/LilyGO-T-Beam#product- ](https://github.com/Xinyuan-LilyGO/LilyGO-T-Beam#product-)
I can find the "TBeamV1.1" on this two lines in the array :
"T-Beam SX1278/76" and "T-Beam SX1262".

Can you please help me to find which SX12## I got on this TBEAM ?

Best regards,

Thierry.

thypl9, 2021-04-06:
I've followed the links for the 3 SX12## in the datasheet part of the REAMDE.MD, here [https://github.com/Xinyuan-LilyGO/LilyGO-T-Beam#datasheet-](https://github.com/Xinyuan-LilyGO/LilyGO-T-Beam#datasheet-)
and on the "Semtech SX1278" page, I can read that this "**78**" is for **137MHz to 525MHz**.

So as the "TTGO" sticker says 868/915Mhz, I think I have a SX1262 or a SX1276...

Can you help me to find which one is on my board ?

Thank you in advance,

Thierry.

lewisxhe, 2021-04-08:
For 868/915Mhz, you can choose SX1276 or SX1262 for testing. If you choose the right, it can run the code smoothly.

If you want to know which type it is, the product title should be clearly marked SX1276/SX1262 in the link of purchase.

thypl9, 2021-04-08:
Thank you very much Lewis !

I was afraid to "brick" my TBEAM by using the wrong firmware... But as you said that the good one can run the code smoothly, and nothing about a risk, I tried both (for my ISM frequency, EU868Mhz).

First, I took note of the COM port number used by my TBEAM in the Arduino IDE (v1.8.13) on MS-Windows.

Then I launched the flash tool, then selected "Developer Mode", then "ESP32 DownloadTool".
I reached the windows that you captured in the picture named "SETTING.png".

I browsed my HDD to reach the folder where I've saved the file "TBEAM_V10_V11_ALLFUNCTION_SX1276_868.bin".

I checked that all the other values where set accordingly to your screen capture "SETTING.png".

I typed the number of my COM port, just above the BAUDs at the bottom right.

I pressed the START button and let it work, till "Finished" appeared.

I restarted my TBEAM and, in the SerialMonitor of the Arduino IDE, at 115200 bauds, got :

"
ets Jun  8 2016 00:22:57
[...]
initBoard
[Radio] Initializing ... **success**!
[...] "

With the other firmware (the file "TBEAM_V10_V11_ALLFUNCTION_SX1262_868.bin"), for the SX1262, I got an "Error code" instead of "success!".

Again, thank you !

closed 2021-04-08 by thypl9

caveman99, 2022-09-26:
Just for the sake of someone finding this via Google - If the T-Beam is 868/915 MHz, and the sticker on the chip only says LORA32 it's a 1276. The other (newer) version has a sticker that reads "SX1262".

lewisxhe, 2022-09-26:
Good suggestion, maybe I should explain how to distinguish LoRa modules