Xinyuan-LilyGO/LilyGo-LoRa-Series #52: No GPS detected: check wiring.

dnwk, 2021-04-13:
I am getting this error using the GPS example with this board  https://www.aliexpress.com/item/4001178678568.html

I tried RX PIN 12 or 34. None of them works.

dnwk, 2021-04-13:
This is TinyGPS example

bosstjann, 2021-04-18:
I have the same problem. Unable to sort out GPS on v1.1. 
GPS led in blinking but not data on serial1.

bosstjann, 2021-04-18:
Hi i have solved my problem with this.
https://github.com/eriktheV-king/TTGO_T-beam_GPS-reset
i needed only version 3.

blub2, 2021-07-14:
Hi I also have two T22_V1.1 devices (20191212 & 20210222) and the GPS examples are also not working.

I tried with @bosstjann solution with V3 & V2 but the examples are still not working.

I flashed the firmware as well https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/firmware
but still no GPS:

> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
initBoard
SparkFun Ublox Example
No Fix - Num. satellites: 0
No Fix - Num. satellites: 0
No Fix - Num. satellites: 0
No Fix - Num. satellites: 0

Siftah, 2022-01-02:
> Hi i have solved my problem with this. https://github.com/eriktheV-king/TTGO_T-beam_GPS-reset i needed only version 3.

Thanks, this also resolved my issue! I could see the GPS light flashing red, so I was fairly sure of a fix, but it needed this running to re-enable the NMEA output!.

fabiano182, 2022-06-24:
Hello everyone,

I have a T22_v1.1 20210222 board and tested all [examples](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/examples) of this repo. The picture is from one of them:

[image]

I've also executed the reset ( https://github.com/eriktheV-king/TTGO_T-beam_GPS-reset  ), without success. I checked all pins voltages and everything looks good. I tested in differents days, indoor and outdoor, no one GPS signal was registred and no blinking on LED.

I think my board is broken. What tests I can do to verify?

Best regards

lewisxhe, 2022-12-08:
There is no activity for a long time. If there is still a problem, please reopen the problem

closed 2022-12-08 by lewisxhe