Xinyuan-LilyGO/LilyGo-LoRa-Series #53: No OLED working

Gonellarro, 2021-04-16:
Hi, 
I recently bought an T-Beam v1.1 with OLED display and after doing everything in the doc, display shows nothing.
I've selected TBEAM board from the Arduino IDE 1.8.3
I've copyed every library inside Arduino libraries folder
I've loaded SSD1306Simple_demo. Set my board in utilities.h and compile everything fine, but nothing appears on the OLED display. What's wrong? Selection of the board?

Thank you in advance.

David-2D3FE9, 2021-04-18:
Hello, 
Did you fix your problem ?

10k-resistor, 2021-05-24:
@Gonellarro Does the OLED have power? You need to to setup the DC1 channel of the AXP192.

Gonellarro, 2021-05-24:
Hello,

Please excuse me that I've not replied yet. Since I wrote you I have not
made any test on the board. I'll try again next july, because I can't
invest time in this project.
I'll write you as soon as I test it again.

Thank you for your concern,
Martí.

Missatge de 10k-resistor ***@***.***> del dia dl., 24 de maig
2021 a les 12:22:

Firered97, 2022-04-13:
hye, i have the same problem, my oled display wont turn on eventhough the code is successfully compiled

lewisxhe, 2022-12-08:
There is no activity for a long time. If there is still a problem, please reopen the problem

closed 2022-12-08 by lewisxhe