Xinyuan-LilyGO/LilyGo-LoRa-Series #56: Cann't acces my sd module with Lilygo TTGO Lora / GPS V1.0 and v1.1 board

far052, 2021-06-30:
Hi all

I buy 2 device. The first one is a  V1.0 with lora and gps neo M6. The second one is a device V1.1 with lora and gps neo m8.

I can read the 2 gps and send / receive lora msg between those 2 device. It was very easy to get/adapt/compile/deploy sample on my device.
I also try bt connection so I can have a big screen to get info. Serial bt Terminal on my phone works very well on the field as long I have battery ;-)

Next step is writing down the nmea msg to a micro sd card.

For this I buy an openlog device (==> https://www.aliexpress.com/item/1005001573018938.html?spm=a2g0s.9042311.0.0.1c496c37pXqnkL)

I want to access to sdcard in spi mode to maximize throughtput as I plan to run the gps in eco mode and send msg every 15 minutes or ... more if possible to maximize battery life.
For this I read the web, git / non git source , try a lot of libraries (sd, sdfat, mysd, etc ..)
But I am not lucky.
I don't understand why it doesn't run.

I try with the sd module running in 3.3V (with usb connected on pc or on battery) and in 5.5V (with pc connected as on battery I only get 3.3v between 5.5V connection and the ground)  with 3x10kohm on the miso line to protect the board

Connection are as follow

> SD Module GND => GND
> SD Mdolule VCC  => 3.3V
> SD Module MISO => GPIO 2 (inspired by some example, also try on gpio 32 as gpio2 may have special meaning/behavior at boot for esp32 device)
> SD Module MOSI => GPIO 15
> SD Module SCK    => GPIO 14
> SD Module RST   => GPIO 13

I try with explicit axp192 initialisation and with no explicit initialisation.
I try with hspi
I try different speed (20mhz, 27mhz, 40mhz and default)

At boot some time the module write a config.txt file and a enpty file LOG00003.TXT (? 3 ?). Not clearly when and why ...
At startup my program stops running with the message "Card failed, or not present". This is cause by the SD.begin call (see my last test sample below). With other sample I get an other msg always resulting from a bad rc from SD.begin call.

Here is a copy of one of the sample I modify for my test. I also join a picture of the 2 llilygo device and the openlog microsd device.
I don't yet make a test with a sd to microsd adapter directly soldered to the lilygo device.

Most of my test have been done with 32GB µcard I bought for this project, As It doesn't run I also make test with an old 512mb and an old 2gb µsdcard. 

Electrical contact has been soldered and seem all ok 

If somebody can help me / give me some advice I will be very happy.

kr

Fred

PS: I have 2 lilygo device. 3 sdcard module. Can't get them running. So I think it is not a hdw pb by my understanding of those device / module

// a sample code

/*
Cette version génère le fichier config.txt comme contenu les 2 lignes suivantes
	9600,26,3,0,1,1,0
	baud,escape,esc#,mode,verb,echo,ignoreRX
et un fichier log LOG00003.TXT vide (0 byte)
*/

#include <SPI.h>
#include <SD.h>

#define I2C_SDA                     21
#define I2C_SCL                     22

#define SDCARD_MOSI                 15
#define SDCARD_MISO                 2
#define SDCARD_SCLK                 14
#define SDCARD_CS                   13
#define SDCARD_SPEED			20000000
#define HAS_SDCARD

#include <string.h>
#define __FILENAME__ (strrchr(__FILE__, '\\') ? strrchr(__FILE__, '\\') + 1 : __FILE__)

#include <axp20x.h>
AXP20X_Class axp;

SPIClass hspi = SPIClass(HSPI);

String dataString =""; // holds the data to be written to the SD card
File sensorData;

void setup() {
  Serial.begin(115200);
  delay(1000);
  Serial.print(__FILENAME__); Serial.println(" is Starting");

  Serial.println("Powering on...");

  //Wire.begin(I2C_SDA, I2C_SCL);
  
  // gpio_pulldown_dis(gpio_num_13);
  // gpio_pullup_en(gpio_num_13);
  
  axp.setPowerOutPut(AXP192_LDO2, AXP202_ON); // lora radio power ?
  axp.setPowerOutPut(AXP192_LDO3, AXP202_ON); // gps main power ?
  axp.setPowerOutPut(AXP192_DCDC2, AXP202_ON);
  axp.setPowerOutPut(AXP192_EXTEN, AXP202_ON);
  axp.setPowerOutPut(AXP192_DCDC1, AXP202_ON);
  delay(2000);
  
  Serial.println("Starting SD");  
  // see if the card is present and can be initialized:
	
  pinMode(SDCARD_SCLK, OUTPUT);
  pinMode(SDCARD_MOSI, OUTPUT);
  pinMode(SDCARD_MISO, INPUT);
  pinMode(SDCARD_CS, OUTPUT);

  hspi.begin(SDCARD_SCLK, SDCARD_MISO, SDCARD_MOSI, SDCARD_CS); //SCLK, MISO, MOSI, SS

  
  //if (!SD.begin(SDCARD_CS, hspi, SDCARD_SPEED)) {
  if (!SD.begin(SDCARD_CS, hspi)) {
    Serial.println("Card failed, or not present");
	  return;
  } 
  else {
    uint32_t cardSize = SD.cardSize() / (1024 * 1024);
    Serial.print("setupSDCard PASS . SIZE = ");
    Serial.print(cardSize);
    Serial.println(" MB");
  }

  if (SD.exists("data.csv")) {
    Serial.println("data.csv exists.");
  }
  else {
    Serial.println("data.csv doesn't exist.");
  }

  // open a new file and immediately close it:
  Serial.println("Creating data.csv...");
  sensorData = SD.open("data.csv", FILE_WRITE);
  sensorData.close();

  // Check to see if the file exists:
  if (SD.exists("data.csv")) {
    Serial.println("data.csv exists.");
  }
  else {
    Serial.println("data.csv doesn't exist.");
  }
  Serial.println("End of test");
  while(1);
}

void loop() {

}

[image]
[image]
[image]

far052, 2021-06-30:
Hi,

I forgot to mention.
First test with bt, lora and gps were made with the board "DOIT ESP32 Dev kit V1" selected in Arduino IDE
As I got trouble with this "board" I also try TTG V1 Board.

Kr,

Fred

LilyGO, 2021-09-11:
1) You need an SD card less than 16GB

2) In order to verify the problem of the library, you can first use the software to simulate the SD card, that is, use other pins, or use this library, https://github.com/nhatuan84/esp32-micro-sdcard or  refer to https://forum.arduino.cc/t/esp32-and-spi-sd-card/658653/3

lewisxhe, 2022-12-08:
There is no activity for a long time. If there is still a problem, please reopen the problem

closed 2022-12-08 by lewisxhe