Xinyuan-LilyGO/LilyGo-LoRa-Series #59: Mesure Volts on pin 35 TTGO 1.6.1

bobo3k, 2021-12-22:
Hi,
I have tried everything (dac, analogRead...) and I never get accurate values out of PIN 35, which should work on a Lilygo T3 V1.6.1
Would you be kind enough to provide a code that allows to get the voltage of the battery (https://www.amazon.es/gp/product/B01KKJP8L6/)?
Many thanks in advance,
Baudouin

vpmayer, 2021-12-29:
Hi, Trie to use the PMU class:
AXP20X_Class PMU;
this library has this function, PMU.getBattVoltage();
but before use you need to init the chip, pmuInit();

bobo3k, 2021-12-30:
Thanks a lot!

I was actually using Pangodream_18650_CL, but it stopped working somehow... I then spotted that the board was faulty, got a replacement, and this same library worked perfectly!

Thaks again!

B

From: Vitor Paganini Mayer 
Sent: Wednesday, December 29, 2021 12:45 PM
To: Xinyuan-LilyGO/LilyGo-LoRa-Series 
Cc: bobo3k ; Author 
Subject: Re: [Xinyuan-LilyGO/LilyGo-LoRa-Series] Mesure Volts on pin 35 TTGO 1.6.1 (Issue #59)

Hi, Trie to use the PMU class:
AXP20X_Class PMU;
this library has this function, PMU.getBattVoltage();
but before use you need to init the chip, pmuInit();

lewisxhe, 2022-12-08:
There is no activity for a long time. If there is still a problem, please reopen the problem

closed 2022-12-08 by lewisxhe