Xinyuan-LilyGO/LilyGo-LoRa-Series #61: Tbeamallfunction deep sleep problem

mattcalavera, 2022-02-25:
on T-Beam v.1.1 when entering in deep sleep mode via longpress all go off, but there is no way to wake up other than leave and reinstert the battery (buttons, reset and others, only make red led on but no more)

netmonk, 2022-02-25:
which firmware are you running ? Or what piece of code ?

mattcalavera, 2022-02-25:
i'm running the "Tbeamallfunction.ino" code (in RadioLibExamples folder)

mattcalavera, 2022-02-25:
https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/examples/RadioLibExamples/TBeamAllFunction

Max-Plastix, 2022-02-25:
Attaching USB power will also wake the PMU.
I too wish there was some way to use a timer to wake the PMU from power-off.

Note that this (longpress OFF) is different from Deep Sleep in the ESP32.
To deepsleep the ESP32, you must keep the PMU (AXP192) on, at least the ESP32 and also OLED power.

I wish the OLED was on a *separate* i2c bus from the PMU, so that the OLED could be powered down as well.

In any case, you are correct -- PMU has no way to power on except battery insertion, power button, or USB power.  There is no timer attached.

ESP32 has more wake-options, but requires keeping the PMU+OLED on at some 5mA power consumption.

lewisxhe, 2022-12-08:
There is no activity for a long time. If there is still a problem, please reopen the problem

closed 2022-12-08 by lewisxhe