Xinyuan-LilyGO/LilyGo-LoRa-Series #62: AXP not detected  in I2C bus

magdalinos, 2022-03-20:
I have a TTGO T-BEAM  T22_V1.0 20190612.
Unit was running fine for more than 2 years. Trying to reset the GPS via scripts found in https://github.com/eriktheV-king/TTGO_T-beam_GPS-reset I am now facing the following issue:

Executing : 

bool initPMU()
{
  Wire.begin(I2C_SDA, I2C_SCL);
  if (PMU.begin(Wire, AXP192_SLAVE_ADDRESS) == AXP_FAIL) {
    Serial.println("AXP_FAIL");
    return false;
  } else {
    Serial.println("AXP_OK");
  }

...

Where
#define I2C_SDA                     21
#define I2C_SCL                     22

I get  the following output 

st:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12804
ho 0 tail 12 room 4
load:0x40080400,len:3032
entry 0x400805e4
initBoard
AXP_FAIL

The unit has a OLED display attached at some point and I am suspecting I run to the issue described in various internet posts about I2C not working properly after deep sleep  or being powered off completely.

Is there any way to restore proper operation and access to the AXP via I2C?

Thank you in advance for your time and support

Max-Plastix, 2022-03-20:
Either the AXP has failed altogether, or failed its i2c communication, or something is disturbing the i2c bus.

Check carefully that no solder bridges or debris somehow messed with pins 21/22 to the OLED, or the OLED power.  You can confirm OLED power with a multimeter after power on.

Do a full power off/on, of course, with no battery attached, using only USB to rule out any strange states.

The issue you mentioned with OLED (powered off) is not related here, since on power-on reset, the AXP is configured correctly.  That issues happens when you tell the OLED controller to turn off, and it then drives the i2c lines in a way that prevents further communication with the AXP until reset. (drags on the data lines)  Reset would clear that.

I also have one device that works like yours, due to my miswiring the OLED at some point.   Surprisingly, the AXP still *works* correctly, it just has no more communication with the ESP32.  So you can talk to the OLED, and the AXP does power the system, but you can't use Button 1 and can't control the AXP from software any more.. device is still useable, but I also get the AXP Fail or AXP Not Found messages.  You might want to try a software build that continues on even after AXP is not-detected, and see what else you can find out .. like whether the OLED is still reachable.

lewisxhe, 2022-12-08:
There is no activity for a long time. If there is still a problem, please reopen the problem

closed 2022-12-08 by lewisxhe