Xinyuan-LilyGO/LilyGo-LoRa-Series #65: T22_v1.1 20210222 configuration

Firered97, 2022-04-15:
i want to ask if i can get any configuration regarding T22_v1.1 20210222 configuration? becoz right now i'm trying to compile the 'gps.ino' with my own modification to make me understand the code a little bit, but there is not result which indicate no gps detected from my t-beam.

void loop() {
  while(Serial1.available() > 0)
    if(gps.encode(Serial1.read()))
    displayInfo();
    

  if(millis() > 5000 && gps.charsProcessed() < 10)
  {
    display.clear();
    display.drawString(20,20, "NO GPS !");
    display.display();
    delay(500);
    while(true);
  }
}

can you help me solve this setting? 
*p/s: the code is working, i just need help with the configuration for the gps as it not show any thing accept for it detect no gps from my t-beam.

netmonk, 2022-04-15:
You have to understand that this board has a power management module controlling power of each element from the voltage source. By default the mcu is powered but the gps module and the radio module aren’t. That’s may be the source of your problem. There are numerous code even in this repo about how to power peripheral by sending commands to this module. 
beware that a little error can happen: asking this power management module to not power anymore the esp32. Which might be a big trouble as far as the standard way to communicate with it is using the esp32

Firered97, 2022-04-15:
@netmonk did you have any guide regarding this situation? becoz before i try this example code.. i have try many example code for this type of t-beam, but nothing is working..this example code is the one that almost working for me than other example code

netmonk, 2022-04-15:
https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/master/examples/GPS/Example1_BasicNMEARead/utilities.h

if you look here you will see an initboard() which call initpmu() (Power Management Unit). 
The PMU is AXP192 and this initboard() function is called in every exemple of this repository.

Firered97, 2022-04-15:
thanks for the guide! 
I will give it a try.

Firered97, 2022-04-15:
do you have other link for the reset gps setup, coz i cant download the firmware tool from that link you give me

Firered97, 2022-04-15:
And also, i have set all the information following utilities.h which is for my board but still same

Firered97, 2022-04-18:
@netmonk hye! i want to thank you for all the help! right now i have get the gps coordinate from the t-beam, even though it require some time to get the coordinate

closed 2022-04-18 by Firered97

netmonk, 2022-04-18:
Great job