Xinyuan-LilyGO/LilyGo-LoRa-Series #66: LoRa RSSI calculation

SaeedMagsi, 2022-04-26:
According to the SX1276 data sheet the RSSI calculation formula should be -157+PacketRssi+PacketSnr*0.25(when using HF port and SNR<0) but here in this code there is no SNR part added. secondly the mid band threshold in the code is 868e6 but it should be 525e6. Am i wrong or is there anything wrong in the code? please let me know because i need a very accurate rssi and snr values for my research. Thank you.

maxgerhardt, 2022-04-27:
[Reference to thread](https://community.platformio.org/t/lora-rssi-calculation-in-liligo-ttgo-lora32-v2-1/273869).

lewisxhe, 2022-12-08:
There is no activity for a long time. Please reopen it if there are other problems

closed 2022-12-08 by lewisxhe