Xinyuan-LilyGO/LilyGo-LoRa-Series #70: Please provide measurements

bzed, 2022-08-04:
Hi,

please provide measurements for you modules, I'm especially interested in the distance and locations of the pin headers on the T-Beam modules. Having to carry around various modules and proper measurement tools is a bit unfortunate when you want to work on projects in your holidays.

Thanks,
Bernd

liangyingy, 2022-08-04:
Is this what you want?

[image]

bzed, 2022-08-04:
@liangyingy This image is mising the distace between the green pin headers columns unfortunately. 
[image]

bzed, 2022-08-04:
(Also the exact location of the pins on the board would be nice, to be able to create an extension baord that with the same screw holes)

liangyingy, 2022-08-04:
[image]

please get it through cad file.  [t-beam_v1.0](../blob/master/cad/t-beam_v1.0.dwg)

bzed, 2022-08-04:
@liangyingy Thank you very much!!

closed 2022-08-05 by liangyingy