Xinyuan-LilyGO/LilyGo-LoRa-Series #75: SX1262 version of TBEAM V1.1?

MrSniffer, 2022-10-03:
Does the schematic here reflect the SX1262 version of the board? Mine has a flashing RGB LED, for example but the markings just say "V1.1". 

And please tell me which arduino.h file corresponds to this board. All the ones I seem to have refer to the SX1276 model.

lewisxhe, 2022-10-03:
The label sticker on the Radio module is 1262 if sx1262 is written, the board can choose TTGO-T-Beam, if only 868MHz is written then it is SX1276

MrSniffer, 2022-10-07:
My sketch needs to use different APIs for a SX1276 board vs. a SX1262 board. What #defines in arduino.h should I use to find out (at compile time) which board I am building for? The Arduino Board Manager doesn't have a "TBeam (SX1276)" and a "TBeam (SX1262)" option, with different #defines in them.

lewisxhe, 2022-10-08:
In fact, you need to define the model used in the radio in the sketch
You can view this:
https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/016cef7be9a475882ff72512e50054e82a53645e/examples/RadioLibExamples/SX1278/SX1278_Receive/SX1278_Receive.ino#L26

MrSniffer, 2022-10-08:
That is exactly what I ***don't*** want to do. There needs to be two different arduino.h files, one for the SX1276 version and one for the SX1262 version, which are automatically set when you select a Board type in the Arduino IDE Board Manager. The source code then uses #ifdef's to compile the right statements.

I'm sorry that you don't comprehend the nature of the problem, Lewis.

Sigh. Someone at LilyGo needs to put some effort into making useful and accurate support information. Another example, the pinout JPG of the SX1262 version is incomplete, and doesn't show BUSY and DIO1.

lewisxhe, 2022-10-08:
You should use ArduinoIDE? I see what you mean. You mean to select the option of adding a Radio on the T-Beam board, right?
I need to confirm the pin diagram with my colleagues, because the pin diagram is not made by me

MrSniffer, 2022-10-08:
Yes, I use Arduino IDE. Having only one Board definition for the two different version of the TBeam V1.1 doesn't permit one to distinguish between the two in the code itself, using some #ifdef etc. preprocessor statements. I think one should select between the two versions using the Board Manager of the IDE, not editing the source code depending on the board.

lewisxhe, 2022-10-08:
This is a good suggestion. At the beginning of T-Beam, there was no clear definition of which radio model to use, so there was only one definition of the board. I agree with your suggestion, but this needs to be addressed to esp32_ Submit board changes in arduinoCore. It will take some time. I will submit PR in these days. Please wait for a while

lewisxhe, 2022-10-08:
[image]
What do you think?
@MrSniffer

MrSniffer, 2022-10-08:
Very nice looking UI. The other part of the job is to make sure there is a corresponding #define in each arduino.h. I'm happy to test a beta version of this when you are ready.

lewisxhe, 2022-10-08:
You can try to replace Arduino first_ Boards in the Core Try the txt file. Remember to back up the original boards.txt first
[boards.txt](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/files/9738502/boards.txt)

[image]

MrSniffer, 2022-10-08:
I don't know what macro to use in my code that will utilize the board and/or variant settings from board.txt. I think the preprocessor defines some ARDUINO_xxxx_yyyy macros for this purpose but I can't find their definitions. Can anyone help out here?

lewisxhe, 2022-10-08:
```

#if defined(TBEAM_USE_RADIO_SX1262)

#elif defined(TBEAM_USE_RADIO_SX1276)

#elif defined(TBEAM_USE_RADIO_SX1278)

#elif defined(TBEAM_USE_RADIO_SX1280)

#elif defined(TBEAM_USE_RADIO_SX1268)

#endif
```

MrSniffer, 2022-10-08:
Aha. OK, now I can hack the build.txt and get the job done! Many thanks for the hand holding.

MrSniffer, 2022-10-08:
I've made edits to my local build.txt and variants folder, but my "revisions" don't appear as part of the Tools menu.  Am I not setting something? Or is the release version from espressif overriding these changes?

t-beam.build.tarch=xtensa
t-beam.build.bootloader_addr=0x1000
t-beam.build.target=esp32
t-beam.build.mcu=esp32
t-beam.build.core=esp32
t-beam.build.board=TTGO

t-beam.menu.Revision.tbeam_V1=TTGO V1.1 LoRa32
t-beam.menu.Revision.tbeam_V1.build.board=TTGO_SX1276
t-beam.menu.Revision.tbeam_V1.build.variant=tbeam-v1
t-beam.menu.Revision.tbeam_V2=TTGO V1.1 SX1262
t-beam.menu.Revision.tbeam_V2.build.board=TTGO_SX1262
t-beam.menu.Revision.tbeam_V2.build.variant=tbeam-v2

lewisxhe, 2022-10-09:
You cannot subdivide variant under tbeam. Please refer to Arduino documentation

https://arduino.github.io/arduino-cli/0.27/platform-specification/

MrSniffer, 2022-10-10:
I've read through that documentation and I don't see anything about Revisions at all! I'll hack a different board, but it is a mystery still why it doesn't work for tbeam. Can you kindly explain the difference between tbeam and ttgo-lora32-oled?

MrSniffer, 2022-10-10:
Can you post or email me your changes to boards.txt that made the five separate board revisions show up? I just can't have any of my changes apply, even by using boards.local.txt.

lewisxhe, 2022-10-10:
I'm not sure which board you mean by ttgo-lara32 oled. The lora of ttgo basically has oled, and there are many versions, including many historical versions. The only difference between them is that the pin distribution is inconsistent. The tbeam is the only version of ttgo esp32 that has GPS and PSRAM, and the other versions have little difference.
Do you want to create different variant files for each board? Why don't you use macro definitions to distinguish between programs?

MrSniffer, 2022-10-10:
Several days ago you posted an image and boards.txt file which showed five different revisions of the basic tbeam board. I am trying to recreate that, and will make my own local edits as needed. But when I install your modified boards.txt in my AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.5 directory, the changes to the Tools menu do NOT take place. Is the Arduino IDE reloading the espressif version anyhow? I am puzzled!

MrSniffer, 2022-10-10:
I finally found the problem and workaround. As indicated in this Issue:
https://github.com/arduino/arduino-ide/issues/1030
the Arduino IDE caches boards.txt in the Windows version. So after editing boards.txt, one has to delete the AppData\Roaming\arduino-ide directory and reload the IDE.

closed 2022-12-08 by lewisxhe