Xinyuan-LilyGO/LilyGo-LoRa-Series #77: TTN_OTAA example fails at T-MOTION device

efreeform, 2022-10-17:
I am trying TTN_OTTA example at the T-MOTION board. I managed to successfully  upload it to the device but after  putting it back  at the USB port at the serial terminal i receive the following:

LoRa Receiver
FAILURE

efreeform, 2022-10-18:
it seems that the PINOUT  for T-MOTION at the example is wrong

efreeform, 2022-10-24:
The PInout is OK, i  added the following line
const lmic_pinmap *pPinMap=&lmic_pins;

and replace os_init(); with
os_init_ex(pPinMap).

Now it passes the os_init state but it fails at the LMIC_reset() with an error  radio.c:1164

lbuque, 2023-02-06:
@efreeform I tested the [LoRaWAN/OTTA](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/examples/LoRaWAN/OTTA) example and it works fine. 

You can update this repo.

I am using platformio, and made some modifications to platformio.ini:

```diff
diff --git a/examples/LoRaWAN/OTTA/platformio.ini b/examples/LoRaWAN/OTTA/platformio.ini
index d8f6050..5b88c43 100644
--- a/examples/LoRaWAN/OTTA/platformio.ini
+++ b/examples/LoRaWAN/OTTA/platformio.ini
@@ -54,5 +54,12 @@ build_flags = -D LILYGO_T3_V2_0
 platform = ststm32
 board = nucleo_l073rz
 framework = arduino
+
 upload_protocol = dfu
-build_flags = -D LILYGO_T_MOTION_S76G
+
+build_flags =
+    -D LILYGO_T_MOTION_S76G
+    -D PIO_FRAMEWORK_ARDUINO_ENABLE_CDC
+    -D USBCON
+    -D HAL_PCD_MODULE_ENABLED
+    -D PIO_FRAMEWORK_ARDUINO_NANOLIB_FLOAT_PRINTF
\ No newline at end of file
```

ttn received the data.

[image]

closed 2023-02-12 by lewisxhe