Xinyuan-LilyGO/LilyGo-LoRa-Series #82: PCF8563_SimpleTime.ino cant write time to device

irqisequalto, 2023-02-01:
running any of the example codes, the time will not set on the PCF8563 device. and reading it spits out noise like:

 Year :2055 Month:0 Day :0 Hour:14 Minute:74 Sec :0
 Year :1955 Month:0 Day :0 Hour:12 Minute:57 Sec :0
 Year :1952 Month:0 Day :0 Hour:16 Minute:40 Sec :0
 Year :1955 Month:0 Day :0 Hour:20 Minute:29 Sec :0
 Year :1952 Month:0 Day :0 Hour:24 Minute:12 Sec :0
 Year :2054 Month:0 Day :0 Hour:21 Minute:81 Sec :0

I have tried this on 2 of the T-Beam-v1.1 devices

lewisxhe, 2023-02-01:
https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/master/README.MD#L113-L113

Just only support s3Version

irqisequalto, 2023-02-01:
So I am unable to access the rtc on the TBeamV1.1 because it's an esp32 device? Is there a rtc on this TBeamV1.1?

lewisxhe, 2023-02-01:
You cannot use this device. There is no RTC on 1.1

irqisequalto, 2023-02-01:
That would be why I can't write to it.. thanks!

closed 2023-02-01 by irqisequalto