Xinyuan-LilyGO/LilyGo-LoRa-Series #83: LilyGo T3S3 V1.0 LoRa SX1280 poor range

palia95, 2023-03-16:
Hello,

In the last months I've been playing a bit with the LilyGo T3S3 V1.0, which has a 2.4GHz SX1280 LoRa modem. 
Anyway, I'm getting a veri poor range on the LoRa protocol (just a few meters) and I really can't understand why, since many people with the same modem can achieve several KMs while performing the ranging.

I am using the same antenna bundled together with the board and I'm using the same ranging example provided by this repo. Any suggestion? I also noticed that a few weeks ago the TXPower limit has been changed from 12 to 3, why? I've been using the old code until last week with the same bad results.

Where am I wrong?

lewisxhe, 2023-03-17:
Are you using the SX1280 with PA?

palia95, 2023-03-17:
I am using the reference example on the Repo

lewisxhe, 2023-03-23:
"No, I mean, do you use the version with PA power amplification? If you are unsure, please send me a product link"

palia95, 2023-03-23:
It's [this one](https://it.aliexpress.com/i/1005004627139838.html?gatewayAdapt=glo2ita), with SX1280 with PA, so yes, it should have it

lewisxhe, 2023-03-23:
oh Please contact the merchant who purchased the board to replace it

closed 2023-03-23 by lewisxhe

palia95, 2023-03-23:
> oh Please contact the merchant you purchased to replace a board. Remember to provide the merchant with a Github conversation record

Ok, but can you please explain me if it's an hardware or a software problem? Because I bought 10 of them. Does the V1.1 change something?

palia95, 2023-03-23:
some of them work up to 3-4 meters, others just a couple of meters, RSSI is very low on all the boards

balynt2048, 2023-04-19:
Hi @palia95 , did you manage to solve this issue?

I have recently purchased 3 pieces of LilyGo T3S3 v1.1 boards with the SX1262 and I seem to have the same issue.
The original arduino firmware on the boards out of the box showed good operation; I set one board to lora send, the other to receive and they could ping with RSSI as good as -33, this lowered to -80 at about 2 meters.
Then, I flashed the board to use micropython and managed to get basic pinging working with some driver I found (https://github.com/ehong-tl/micropySX126X). But now I see the what you are seeing, that even when the boards are at 1 cm away, the best RSSI I get is -80, this decreases to -110 at 1 meter, then about 2 meters, the signal is lost completely.

My guess is that there is something wrong with how I program the SX chip to configure the power amplifier, even though there really is a command to set power settings in the initialization sequence of my micropython driver.

Would be grateful for any tips! Thanks!

palia95, 2023-05-13:
> 

Hello, I got some new boards and even the new ones have the same issue of the poor lora range. I'm using the last example code available on GitHub

lewisxhe, 2023-05-13:
Which sample code did you use?

palia95, 2023-05-13:
> Which sample code did you use?

[THIS](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/examples/temporary)

Master and Slaves

But even using the old one, with TXpower now set to 3, I still get a poor range

lewisxhe, 2023-05-17:
This was tested by another person, and I only uploaded it to T3S3 users for testing. Recently, I was on vacation. Now I have tested that this is problematic and has no reference value.
Can you test [this](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/examples/RadioLibExamples/SX1280 ) It should be normal
Regarding distance measurement, I am still actively communicating with the manufacturer to ask them to help us solve this problem. I will update them if there is progress in the future