Xinyuan-LilyGO/LilyGo-LoRa-Series #85: libraries for examples do not work

warnerthuis, 2023-03-24:
I have the LilyGO TTGO T-Beam V1.1 an installed the libraries in C:\Users\jpfla\Documents\Arduino\libraries as indicated.
Then I get the following error:
In file included from C:\Users\jpfla\AppData\Local\Temp\arduino_modified_sketch_546438\TinyGPS_Example.ino:7:
boards.h:26:10: fatal error: XPowersAXP2101.tpp: No such file or directory
 #include "XPowersAXP2101.tpp"
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
I get this error with Arduino IDE 1.8.19 as well as with 2.0.4 and with more examples than this one e.g. TinyGPS_KitchenSink:
In file included from C:\Users\jpfla\AppData\Local\Temp\arduino_modified_sketch_825818\TinyGPS_KitchenSink.ino:7:
boards.h:26:10: fatal error: XPowersAXP2101.tpp: No such file or directory
 #include "XPowersAXP2101.tpp"
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.

lewisxhe, 2023-03-25:
Please re clone the repository and re copy the XPowersLib in the lib folder to your arduino libraries directory

warnerthuis, 2023-03-25:
Done. Nothing changed.

warnerthuis, 2023-03-25:
I'll try on a different pc that is running W10 in stead of W11.

lewisxhe, 2023-03-25:
请你查看boards.h 中是否与现在的一致!
https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/fea8c3397aa97ffb2e477bbfcfeb9f7d34f8c8d5/examples/GPS/TinyGPS_KitchenSink/boards.h#L26

warnerthuis, 2023-03-25:
on the other machine it works ok. So I will (after breakfast) on my main machine remove both ide's and do a clean install.
But first breakfast.

lewisxhe, 2023-03-25:
No, in fact, you don't need to delete Arduino, you just need to delete `Lilygo-Lora-Series` this folder, then replace the Xpowerslib in libraries re -replaced with the current Xpowerslib

warnerthuis, 2023-03-25:
To be certain I rmoved both ide's from the pc and deleted all directories also in AppData.
Now it works.

closed 2023-03-25 by warnerthuis