Xinyuan-LilyGO/LilyGo-LoRa-Series #87: axp192 reset

Marcotic, 2023-03-28:
Hi. I was having problems with the battery and the axp192 not charging and by mistake i uploaded a code where i set dcd1 voltage to 1.2v.
Now my tbeam v1.1 is not booting anymore. Red led lights on boot but immediately shuts down. I cannot upload anymore. 
Is there a way to reset axp192. Or reflashing the board without using usb?
Thanks

lewisxhe, 2023-03-28:
1. Discount the power supply of USB and batteries
2. Connect GPIO0 and GND through the wires
3. Insert USB
4. Upload the right sketch
5. After the upload is completed, open the USB and connect again

Marcotic, 2023-03-28:
THank you so much!!!!

I was able to upload again.

I still have the issue of the battery. Tbeam wont run with battery only. Battery is fully charged. It wont charge it neither. 
Any advice

lewisxhe, 2023-03-28:
[image]
Do you have a multimeter? The voltage at these two points can be measured See if it's normal

Marcotic, 2023-03-28:
Measured 4.05 volts.

lewisxhe, 2023-03-28:
Did you press the PWR button for two seconds?

Marcotic, 2023-03-28:
Uploaded LoRa sender code form the repo, measured before  pressing 4.04, measured after pressing 2sc same 4.04.

lewisxhe, 2023-03-28:
Can I turn off the board power supply by pressing and holding the PWR button if I use USB power supply?

lewisxhe, 2023-03-29:
[image]
Then measure whether there is voltage at these two points

Marcotic, 2023-03-29:
voltage of the last two points:
usb powered: Normal 4.88 after 8 sc holding power 4.95
on battery powered..no red led, 1.32

I just noticed that measuring into first two the points the ones from the https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/issues/87#issuecomment-1486515120 image, it has 3.85volts. I f it helps...

Marcotic, 2023-03-31:
any idea??

lewisxhe, 2023-03-31:
Sorry, I forgot to reply to you
1. PWR can be used for power-on and power-off operations when using USB
2. Measure the voltage at two points of the MOS tube in the picture
3. Unable to start the machine when using the battery alone, (press and hold the PWR button for two seconds)
Please confirm the above precautions. If all conditions are met, the AXP192 may be damaged