Xinyuan-LilyGO/LilyGo-LoRa-Series #90: LilyGo T3 v1.8 "Timed out while waiting for complete SPI command"

Llbes98, 2023-04-24:
Hi i'm doing a project where i want to test LoRa 2.4Ghz versus the sub Ghz LoRa.
For this i use the LilyGo T3 v1.8 to test LoRa 2.4Ghz, however i am having problems getting it to work. Im using the RadioLib examples, specifically the SX128x_Transmit_Interrupt.
I'm pretty sure i have installed all the required libraries correctly, and i have uncommented line 12 where the LilyGo T3 v1.8 is defined in the utilities.h file. Other than that i have not altered the code at all, however i keep getting this output on the serial monitor:

"
initBoard
Started OLED
setupSDCard FAIL
[SX1280] Initializing ... failed, code -705
"

I've been searching for the error code, and according to the RadioLib documentation found here: https://jgromes.github.io/RadioLib/group__status__codes.html
it means that "[SX126x](https://jgromes.github.io/RadioLib/class_s_x126x.html) timed out while waiting for complete SPI command."
which i find weird since im using the SX1280 chip, but i also haven't been able to find any help on how to fix it. can anyone help?

lewisxhe, 2023-04-24:
https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/firmware/hardware_test/T3_V1.8_2.4G

Llbes98, 2023-04-25:
> Burn the factory firmware to quickly verify the problem firmware: https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/firmware/hardware_test/T3_V1.8_2.4G flash method: https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/firmware

Thanks for helping.
I have tried this now, and when i use the method provided in your answer everything works. However i might just not know enough, since im still not really able to verify the problem... If i try to upload code via the Arduino IDE it still doesnt work (same error).

From the flash download tool i get this information after flashing to the MC:
> AP:  4C7525A78CD9  STA:  4C7525A78CD8
BT:  4C7525A78CDA  ETHERNET:  4C7525A78CDB

And "Detected Info":
> flash vendor: C8h : GD
flash devID: 4016h
QUAD;32Mbit crystal: 40Mhz

Im adding the code im working with below i might have done something wrong, (again its the example code provided here, but i guess i could have done something wrong with it)

again thanks for your help.

SX128x_Transmit_Interrupt:
```c++
/*
   RadioLib SX128x Transmit with Interrupts Example
   This example transmits LoRa packets with one second delays
   between them. Each packet contains up to 256 bytes
   of data, in the form of:
    - Arduino String
    - null-terminated char array (C-string)
    - arbitrary binary data (byte array)
   Other modules from SX128x family can also be used.
   For default module settings, see the wiki page
   https://github.com/jgromes/RadioLib/wiki/Default-configuration#sx128x---lora-modem
   For full API reference, see the GitHub Pages
   https://jgromes.github.io/RadioLib/
*/

#include <RadioLib.h>
#include "boards.h"

SX1280 radio = new Module(RADIO_CS_PIN, RADIO_DIO1_PIN, RADIO_RST_PIN, RADIO_BUSY_PIN);

// save transmission state between loops
int transmissionState = RADIOLIB_ERR_NONE;

// flag to indicate that a packet was sent
volatile bool transmittedFlag = false;

// disable interrupt when it's not needed
volatile bool enableInterrupt = true;

uint32_t counter = 0;

// this function is called when a complete packet
// is transmitted by the module
// IMPORTANT: this function MUST be 'void' type
//            and MUST NOT have any arguments!
void setFlag(void)
{
    // check if the interrupt is enabled
    if (!enableInterrupt) {
        return;
    }

    // we sent a packet, set the flag
    transmittedFlag = true;
}

void setup()
{
    initBoard();
    // When the power is turned on, a delay is required.
[... 513 lines cut ...]
        }

    } else {
        uint32_t cardSize = SD.cardSize() / (1024 * 1024);
        if (u8g2) {
            do {
                u8g2->setCursor(0, 16);
                u8g2->print( "SDCard:");;
                u8g2->print(cardSize / 1024.0);;
                u8g2->println(" GB");;
            } while ( u8g2->nextPage() );
        }

        Serial.print("setupSDCard PASS . SIZE = ");
        Serial.print(cardSize / 1024.0);
        Serial.println(" GB");
    }
    if (u8g2) {
        u8g2->sendBuffer();
    }
    delay(3000);
#endif

}
```

lewisxhe, 2023-04-25:
You mean if it is normal to write to the bin file, but it cannot be used when using the sketch, right?

lewisxhe, 2023-04-25:
From the pin definition of the schematic and sketch there is no error, I am on vacation now, so I can't verify it, I will test it in mid-May.

Llbes98, 2023-04-25:
Allright, thank you very much for your help. I will try to figure it out myself, if i find the problem i'll write here, otherwise i would still be very happy if you figure it out :)

lewisxhe, 2023-04-28:
I have found the test code, but I am currently unable to verify if it is working properly. Please give it a try, I will update the relevant examples after the holiday
[T3 V1.0 Factory]（ https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/examples/temporary )
If it's feasible, please let me know

lewisxhe, 2023-05-13:
[SX1280](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/examples/RadioLibExamples/SX1280) Transmission and reception examples are currently available, but the ranging example is not accurate, I It has been removed. I don't have much information about distance measurement at present, I will test it later

closed 2023-05-13 by lewisxhe