Xinyuan-LilyGO/LilyGo-LoRa-Series #91: Ranging example crashes after 10 missed reads on T3S3

mjaakkol, 2023-04-26:
Ranging example ranging master is crashing after following traces:

NotValid,Irq,400
Start Ranging

appear 10 times. I managed to make things a bit more stable by moving sleep outside if so that it is happening each loop and not just in successful ones but once the device is out of the range, the device gets rebooted after 10 tries. 

Btw. Does anyone know what kind of range can one expect with the out-of-box device? I'm not able to get more than ~10m.

balynt2048, 2023-04-26:
Is this with the SX1280 LoRa chip? There was a related issue [here](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/issues/83).

mjaakkol, 2023-04-26:
Yes, this is SX1280 and they are kind of related but not the same. The fact that PA seems to be broken in these units just make the issue with no-signal easier to reproduce.

lewisxhe, 2023-04-27:
Write the factory test firmware to check whether the distance is normal, so that it can be judged that the PA is burned due to the wrong setting of the PA power.
https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/master/firmware/hardware_test/H594_LoRaS3_1.0/T3_S3_V10_SX1280.bin

If the distance is still very short after writing, then there is no doubt that the PA of the board has been burned and cannot be repaired.

mjaakkol, 2023-04-27:
With test firmware I'm able to keep up the connection at least the double the distance and having -85dBms while the ranging example would have half the distance with dBm value -105. It starts to look that HW is OK but ranging sample is not using PA. Is this something one needs to enable in SW themselves? If so, let me know what to try.

lewisxhe, 2023-04-27:
Sorry, I can't answer what you need to set for this, because T3 S3 is not tested by me, I need some time to verify, I am currently on vacation, I will probably test in mid-May

mjaakkol, 2023-04-27:
Are you able to share the test-firmware source code so that I could take a look if the current code is missing some initialization step?

lewisxhe, 2023-05-13:
I tested the ranging and there was a problem. Although I found the cause of the crash, the results I obtained were inaccurate. I have deleted it, and currently sending and receiving are normal. I will test the ranging in the later stage.

mjaakkol, 2023-05-13:
Thanks for looking into this. Do you have any estimate when you have tested the ranging? I'm hoping to your hardware on a demo and ranging is the feature that I need the most while, of course, the data connectivity needs to work as well.

palia95, 2023-05-15:
Hello, I confirm I am having the same issue, very poor distance in ranging mode with the SX1280 modem

lewisxhe, 2023-05-15:
At present, I am using <semtech sx1280> and found that there is also a problem with the ranging example. I think it may be a problem with the PA. Please give me some time. I am applying for the SX1280 without PA for testing

lewisxhe, 2023-05-17:
Hi, everyone
Can you test this(https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/examples/RadioLibExamples/SX1280 ) 
Their distance should be normal
Regarding distance measurement, I am still actively communicating with the manufacturer to ask them to help us solve this problem. I will update them if there is progress in the future

mjaakkol, 2023-05-17:
I think I already mentioned that the standard RX & TX stuff works just fine regarding the range that makes me wonder what's wrong with the ranging side. While you were on vacation, I tried to just use RadioLib for ranging in hope that the library that you used for ranging (different from TX and RX samples) would be working but it didn't. It looks like modem is turning PA off when doing the ranging (cannot confirm as I don't have spectrum analyzer in my use).

palia95, 2023-05-18:
I am almost sure that in ranging mode the PA is off. With the standard RX and TX example and the boards set at 50 cm of distance, the RSSI of the last LoRa messages is around -18/-20 dBm, while in ranging mode at the same distance it's around -80 dBm, I'm using the same TXPower setting set to 3, therefore something is wrong with the power settings (probably as you already stated, PA is off). Thanks anyway for the effort

mjaakkol, 2023-05-18:
I would assume that modem firmware fix to control PA should work for this. This probably cannot be done through Arduino code, right?

lewisxhe, 2023-05-23:
I put my test code on my github, you can try it. But I am busy with other things recently, I will continue to try after a while

[SX1280Ranging](https://github.com/lewisxhe/SX1280Ranging)

lewisxhe, 2023-05-23:
I just got the original development board, and their test results are not satisfactory. I don't think it is necessary to study the distance measurement function of SX1280. It is very inaccurate and has no reference value.
[image]

lewisxhe, 2023-05-23:
[image]

palia95, 2023-05-23:
Hello, thank you for all your tests and effort. Before buying these boards, I investigated the performance of the SX1280 ranging capabilities and you can easily find several papers online which exploited them to perform trilateration with a few meters error accuracy (which is reasonable for such frequency, given the spatial resolution). I would suggest to test the ranging over a much longer distance, since as reported by the aforementioned papers that you can find attached, the system is more suitable for long range applications (more than 100 meters, where a few meters error is accettabile).
It would be nice to understand how these people didn't have any issue with the raging feature of the SX1280, since they managed to cover several km. Maybe a different version of RadioLib?

[REF 1](https://stuartsprojects.github.io/2019/04/26/Semtech-SX1280-2-4Ghz-LoRa-ranging-tranceivers.html)
[REF 2](https://ieeexplore.ieee.org/document/9221049)
[REF 3](https://github.com/alphaLeporis/LoRa_2G4_localization/blob/main/documentation/LoRa_localization_analysis.pdf)

mjaakkol, 2023-05-23:
I think there are multiple things in play here.
SX1280 ranging is based on the time-of-flight and short distances would require extremely accurate clocks and E2E sampling pipeline so basing the system quality around few meter distance is not representative. Having said that when I tested the system in 5-8m range, it was pretty accurate and if I would have been able to go further, I think the error % compared to measured distance would have dropped even further. This makes me conclude that ranging itself works like with any SX1280.

The real issue is that range. These should have 1-2km range even with 2.4GHz and I'm getting 10m with ranging. There is something wrong how PA is operated. I ported the ranging examples from RadioLib into HW and didn't make a difference so the root-cause is somewhere deeper. If I would have to bet, something at radio modem layer is not turning PA on when doing ranging but does so when operated in normal mode. I assume this to be problem for this particular HW as @palia95 noted, the chipset has been tested successfully with other devices.

lewisxhe, 2023-05-24:
I added PA switch for SX1280 ranging in RadioLib https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/commit/229ad6f5bf93f6fc473290be7f05f89120d135d3

And uploaded the test example that I modified and tested, so far I can only do this. 
[Ranging](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/examples/RadioLibExamples/SX1280/Ranging)

mijaakkola, 2023-05-25:
I tried to get the example using RadioLib to work but Master is just reading timeout. Only code change that I made was to uncomment the board so that it is compiled for this board. Any ideas what I did wrong?

I also tried to compile the other sample but I don't find #include <SX128XLT.h> from the codebase. Was this removed?

lewisxhe, 2023-05-26:
@mijaakkola The actual library has been updated, please try again

mijaakkola, 2023-05-27:
I tested the new SX128LT based ranging and ranging itself seems to be there but the range is even worse than before. Just walking few meters out, will drop the connection. The starting dBm is like -110 and drops quickly into -120 losing the connection.

I didn't get the ranging to work at all with RadioLib based sample as it could just show timeout all the time.

Did you manage to get ranging to operate over long range (100m+)?

palia95, 2023-06-05:
Hi, I tried the last update and I still have the poor range in ranging mode. Should I suppose that there isn't going to be a fix? Did someone tried on another board with the SX1280 if it work? Thanks

mijaakkola, 2023-06-05:
This works on SX1280 reference board so the range related issue is specific to this hardware.

nbeirante, 2023-09-12:
Hi!!  I have the same problem with the examples and a T3S3 v1.2 sx1262 board. :( Kind regards.

closed 2024-04-12 by github-actions[bot]