Xinyuan-LilyGO/LilyGo-LoRa-Series #92: LoRa and the factory default firmware

balynt2048, 2023-04-26:
Hello devs, dear @lewisxhe,
I would like to ask for more information about the default firmware the boards are shipped out with. I need this help because I have a  [problem](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/issues/83#issuecomment-1514575428) with using the LoRa module on my LilyGo T3S3 v1.1 boards (SX1262 LoRa chips model). 

Briefly, my problem is that I can't get LoRa to work properly by any means other than the default firmware. I intend to use micropython on the ESP32, and tinkering with [this driver](https://github.com/ehong-tl/micropySX126X), I managed to get ping sending and receiving to work. (This appears to be a python reworking of the RadioLib SX1262 and SX126x module. I read both and they _should_ be equivalent. The maintainer of the driver unfortunately [could not help](https://github.com/ehong-tl/micropySX126X/issues/17)). However, there seems to be a problem with setting the transmission power, because the working range is a few (2-3) meters only, no matter how I try to set power-related settings and registers, it does not change.

I also tried the examples given in this present repo, but unfortunately I was unable to make those work at all (I might open a separate issue for that, I think I put in the wrong pin numbers).
Basically, the only way I can get to see good operation of the LoRa (good enough range, transmission through walls etc.) is the factory firmware. I managed to save the original flash contents via read_flash and saw @lewisxhe 's name pop up in the binary, so I figured you or your team should be able to help out. My questions specifically:
- Is is possible to see the source code of the factory default firmware? I failed at decompiling.
If not, then:
- Could you tell me which radio lib does the default firmware use, and with what settings (bandwith, coding rate, spreading factor, private/public keyword, power (dBm))? My guess is that the LoRa ping-pong parts of the firmware is based on one of the examples (I see the same "[SX1262] Received packet!" prints in the RadioLib receive interrupt example that the firmware on my board is printing on the serial port).

Thank you for any help in advance!

lewisxhe, 2023-04-27:
T3 S3 This is not tested by me, I need to compile the binary file and send it to LilyGo for testing to verify that the source code is normal.
The default firmware is to use RadioLib, I am sure of this

balynt2048, 2023-04-27:
Hello @lewisxhe, thanks for getting back to me. If you are the person compiling the binary, could you check which pins are used in the RadioLib module constructor? Also, could you tell me the transmission parameters, like frequency, bandwith, spreading factor, coding rate?

I've just opened a separate issue about using the RadioLib examples #93.

lewisxhe, 2023-04-27:
This warehouse was built and maintained by me, but the T3 S3 version was not submitted by me, it came from another submitter, I need time to verify this problem, because I am currently on vacation and I don’t have a board , so can't verify, I can only send binary file to LilyGo for verification. Please give me some time

lewisxhe, 2023-04-28:
I have found the testing code for, but I am currently unable to verify if it is working properly. Please try it out and I will update the relevant examples after my vacation
[ T3 V1.0 factory ](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/examples/temporary/sx1262_Integration)

balynt2048, 2023-04-28:
Thank you making it available. I managed to build and burn it onto my two board. Unfortunately, they did not work regarding the LoRa transmitting-receiving. I have to try out some more settings about the PINS. Will let you know how it goes.

lewisxhe, 2023-05-13:
I have made modifications to the error section and it is now usable. [SX1262](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/examples/RadioLibExamples/SX1262)
Regarding the factory firmware, although it can run, there are errors. I will organize and submit it later

closed 2023-05-13 by lewisxhe