Xinyuan-LilyGO/LilyGo-LoRa-Series #93: Cannot make LoRa examples work (T3S3 v1.1 board, SX1262 chip)

balynt2048, 2023-04-27:
I tried using the RadioLib examples for the SX1262 transmit interrupt and receive interrupt without success.
I think there is a bug in the [utilities.h](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/5183adc6332b94cc3097b8eb6f8aff042b70f61b/examples/RadioLibExamples/SX1262/SX1262_Receive_Interrupt/utilities.h#L237) regarding the T3S3 v1.0 definitions: 
#elif defined(LILYGO_T3_S3_V1_0)
...
#define RADIO_SCLK_PIN              5
#define RADIO_MISO_PIN              3
#define RADIO_MOSI_PIN              6
#define RADIO_CS_PIN                7
// #define RADIO_DIO0_PIN              9
#define RADIO_TCXO_EN_PIN           33
#define RADIO_DIO1_PIN              33
#define RADIO_BUSY_PIN              36
#define RADIO_RST_PIN               8
#define RADIO_DIO2_PIN              34
#define RADIO_DIO5_PIN              36
#define RADIO_RX_PIN                21
#define RADIO_TX_PIN                10

The RadioLib uses the RADIO_DIO1_PIN as the GPIO pin [in the example:](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/5183adc6332b94cc3097b8eb6f8aff042b70f61b/examples/RadioLibExamples/SX1262/SX1262_Receive_Interrupt/SX1262_Receive_Interrupt.ino#L23)
The constructor should look like this:
new Module(chip select pin, interrupt pint, reset pin, gpio pin).
`SX1262 radio = new Module(RADIO_CS_PIN, RADIO_DIO1_PIN, RADIO_RST_PIN, RADIO_BUSY_PIN);`

But on this board, GPIO33 is used as an enable pin for the TCXO on the radio module. It is always set to [high](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/5183adc6332b94cc3097b8eb6f8aff042b70f61b/examples/RadioLibExamples/SX1262/SX1262_Receive_Interrupt/boards.h#L321). Therefore, there should be another pin defined here as the main GPIO pin. 9 is a good candidate, but it is commented out here? Anyways, I tried using GPIO 9 as the GPIO pin for the radio module (and I tried many other pins, too), but no success.

Could someone explain to me what the interrupt and gpio pins should be set as, especially on the T3S3 v1.1 board? Thanks in advance!

lewisxhe, 2023-04-27:
as mentioned here.
https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/issues/92#issuecomment-1525539442

lewisxhe, 2023-04-28:
I have found the testing code for, but I am currently unable to verify if it is working properly. Please try it out and I will update the relevant examples after my vacation
[ T3 V1.0 factory ](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/examples/temporary/sx1262_Integration)

lewisxhe, 2023-05-13:
I have made modifications to the error section and it is now usable. [SX1262](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/examples/RadioLibExamples/SX1262)

closed 2023-05-13 by lewisxhe