Xinyuan-LilyGO/LilyGo-LoRa-Series #95: T-Beam Supreme Arduino IDE Settings for SoftRF?

VirusPilot, 2023-05-29:
I am trying to compile/flash SoftRF for my just arrived T-Beam Supreme board in Arduino IDE but maybe some of my settings are incorrect so that the final flashing stage won‘t work. SoftRF hasn‘t published any settings either so I am just guessing about the settings - please advise (Board: ESP32S3 Dev Module?):
- USB CDC on Boot (enabled?)
- USB DFU on Boot (disabled?)
- Flash Mode (QIO or OPI?)
- Flash Size (8MB with spiffs?)
- USB Firmware MSC on Boot (disabled?)
- PSRAM (OPI PSRAM?)
- Upload Mode (UART0 / Hardware CDC?)
- USB Mode (Hardware CDC and JTAG?)

lewisxhe, 2023-05-29:
There are firmware build instructions here.
The key points are not explained, I make the following explanations
```
USB CDC on Boot (enabled)
USB DFU on Boot (disabled)
USB Firmware MSC on Boot (enable)
PSRAM (QSPI PSRAM)
Upload Mode (UART0 / Hardware CDC)
USB Mode (Hardware CDC and JTAG)
Flash Mode (QIO)
Flash Size (8MB with spiffs) ESP32S3 uses tinyuf2 partition table, https://github.com/adafruit/tinyuf2/blob/master/ports/espressif/partitions-8MB.csv You need to rename `partitions-8MB.csv` to esp32 `default_8MB.csv` in the core tools\partitions directory
```

@lyusupov Am I right?

VirusPilot, 2023-05-29:
Thanks for now, will try tomorrow.

VirusPilot, 2023-05-30:
With "USB CDC on Boot (enabled)" I am running into the following error:

../SoftRF/software/firmware/source/SoftRF/src/platform/ESP32.cpp:1047:7: error: 'USB' was not declared in this scope
   if (USB.manufacturerName(ESP32SX_Device_Manufacturer)) {

VirusPilot, 2023-06-06:
So as mentioned, all fine now since USB-OTG CDC (TinyUSB) is selected.

closed 2023-06-06 by VirusPilot