Xinyuan-LilyGO/LilyGo-LoRa-Series #102: WiFi ipx connector

dodo124, 2023-09-11:
Hi, which resistor has to be removed when using the ipx connector? I can't find the information.

Thanks Dominik

balynt2048, 2023-09-21:
Well first please specify which board you have.

dodo124, 2023-09-21:
Oh my bad. Sorry. I do have the following board: GO®
LoRa32 V2.1_1.6. Regards Dominik

balynt2048, 2023-09-21:
Are you sure it is this https://www.aliexpress.com/i/32872078587.html ? What does it say on the board? I don't see a second ipex connector for the wifi, only one for the LoRa antenna.
If you look at for example the T3S3 v1.1 board (which I have right now), there are two ipex connectors.

dodo124, 2023-09-21:
Hi,
Yes it's the board and it only has one ipx like it is shown in the picture. So I guess there is no way to attach a ipx to the board. The problem is that the onboard WLAN antenna has fallen off. So I can't turn on WiFi. Any suggestions in my particular case?

Regards Dominik

closed 2024-04-12 by github-actions[bot]